Ultimaker/Cura #8157: Combine Infill Areas when Infill Matches Top/Bottom

CCS86, 2020-08-06:
What do you guys think about combining infill areas when the "Top/Bottom Pattern" matches "Infill Pattern" (most commonly 100% lines).

I have noticed that on some models a large amount of additional travel and retraction is added to keep portions of a layer which are top/bottom compartmentalized from normal solid infill. 

Would it be difficult to see that the patterns and density match, and simply recombine the areas and fill as one?

Ghostkeeper, 2020-08-10:
Then it also needs to match:
* Infill Extruder == Top/Bottom Extruder
* Infill Line Directions == Top/Bottom Line Directions
* Extra Infill Wall Count == Extra Skin Wall Count
* Infill Overlap == Skin Overlap
* Infill Line Width == Top/Bottom Line Width
* Infill Speed == Top/Bottom Speed
* Infill Acceleration == Top/Bottom Acceleration
* Infill Jerk == Top/Bottom Jerk
* Infill Flow == Top/Bottom Flow
* (Infill Pattern == Top/Bottom Pattern, as you said)
* (Infill Line Distance == Infill Line Width, as you said)

And some features need to be disabled, to allow skin and infill to print the same:
* Connect Infill Lines == False
* Infill X Offset == 0
* Infill Y Offset == 0
* Randomize Infill Start == False
* Infill Wipe Distance == 0
* Infill Layer Thickness == Layer Height
* Gradual Infill Steps == 0
* Infill Before Walls == False
* Infill Support == False
* Combing Mode != Not in Skin

(The following settings would be ignored because they don't matter if the infill is the same as the skin:)
* Infill Line Multiplier
* Minimum Infill Area
* Skin Removal Width
* Skin Expand Distance
* Maximum Skin Angle for Expansion
* Skin Edge Support Thickness

CuraEngine could check for all of these things and if so, pretend that everything is skin. Doing so would improve print quality of completely solid parts since it won't interrupt the fill lines halfway.

However this case is quite rare. Most profiles have some degree of infill overlap and infill wipe distance, or infill before walls enabled. If those differences are ignored, and it's all implemented in CuraEngine, there would be no way to make it properly listen to these settings any more at 100% density.

That's why we haven't implemented this in CuraEngine but in the front-end and ignore most of these additional settings; only the density is used. If printing at 100% density, it should combine everything to be Top/Bottom. This way there is a way to undo the change if those settings are important; something that would be a bit advanced for a user to do but at least a profile author should be able to do.

CCS86, 2020-08-17:
That makes a lot of sense. Not a huge gain for all the complication.

closed 2020-08-17 by CCS86