Ultimaker/Cura #8160: Not detected OpenGL when load from Microsoft Remote Desktop

humbertosales, 2020-08-07:
**Application version**
4.6.2

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Connect in remote computer using Remote Desktop (RDP)
2. Start Cura
3. Message show.

**Screenshot(s)**
[image]

**Actual results**
Computer with OpenGL, but Cura not start because not detect OpenGL via RDP.

**Expected results**
Start Cura.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Mosher23, 2020-08-08:
its not cura its your pc, any program which uses OpenGL will not start over RDP

fieldOfView, 2020-08-08:
If you happen to be using an NVidia GPU, there's a patch to fix the problem:
https://www.khronos.org/news/permalink/nvidia-provides-opengl-accelerated-remote-desktop-for-geforce-5e88fc2035e342.98417181
If not, you are out of luck. It is your OS telling Cura (or other applications that need OpenGL) that there is no GPU in the system.

Ghostkeeper, 2020-08-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/5474.

closed 2020-08-10 by Ghostkeeper