Ultimaker/Cura #8163: build plate temperature initial layer doesn't work.

lmaxl04, 2020-08-08:
**Application version**
4.6.2

**Platform**
windows 10

**Printer**
ender 3

**Reproduction steps**
1. build plate temperature : 0 (any value)
2. build plate temperature initial layer : 40
3. slice
4. save G-code

**Screenshot(s)**
N/A

**Actual results**
build plate temperature initial layer : 0
after printing initial layer, build plate temperature : 0

**Expected results**
build plate temperature initial layer : 40
after printing initial layer, build plate temperature : 0

**Project file**
N/A
**Log file**
N/A

**Additional information**
N/A

jellespijker, 2020-08-09:
I can't reproduce your issue.
When I set the `build plate temperature` to 1 (0 will leave it unchanged as is shown in the tooltip) and the `build plate temperature initial layer` to 40 the resulting GCode Shows and `M140 S40` at the start and a `M140 S1` at the start of layer 1

My GCode 
```GCode
;FLAVOR:Marlin
;TIME:16267
;Filament used: 9.44242m
;Layer height: 0.2
;MINX:66.849
;MINY:82.733
;MINZ:0.2
;MAXX:168.148
;MAXY:152.256
;MAXZ:98.4
;Generated with Cura_SteamEngine 4.6.2
M140 S40
...

...
;LAYER:1
M140 S1
...
```
Can you send us your log and project file?

vovodroid, 2022-09-23:
> 0 will leave it unchanged as is shown in the tooltip

Tooltip of "`build plate temperature`" in version 5.1.0 says "`if set to 0 plate is left unheated`". If I set it to 0 and initial layer to 80 I get only M140 S80 in beginning.  But if I set `build plate temperature` to any value just above 0, like 0.01 there is `M140 S0` **before** first layer

```
;LAYER:1
M140 S0
```
I would expect turn the plate off **after** first layer.