Ultimaker/Cura #8166: Allow Prime Tower setting height.(use 2-in-1 hotend)

sw35, 2020-08-10:
Printer
Custom Printer , 2-in-1 Hothend ,2 extruders with shared heater(mixed nozzle).

**Is your feature request related to a problem? Please describe.**
When I print bigger things, because Prime Tower not be printed to the top, 
there will always be problems,  Ex. material liquefaction, dripping, dyeing

I think it’s because another material has been in the heating zone for too long, causing problems

[image]

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
I always create a new square and designate it as a support material,
The height is the same as the model, and it is forced to be printed on top.
To solve this problem

But this is a waste of material and time.

Hope to increase the option of setting Prime Tower height

[image]

Sorry I am not aware of how to fill an issue and  English is not my native language

Ghostkeeper, 2020-08-11:
What happens when you make a single-extrusion print? Does it also cause liquefaction, dripping, discolouration?

The prime tower only lasts until the last extrusion switch, because after that, the print is basically just a single extrusion print. To the nozzle it doesn't matter whether it continues printing on one big layer, or whether it continues with a higher layer. It just keeps on printing.

If your problem is actually that the filament degrades when it's heated for too long, I suggest you lower the printing temperature. All 3D printed filament is designed to be heated while in the Hot Zone of the nozzle of a normal continuous print.
If your problem is that the print is too fast and the previous layer hasn't solidified yet, I suggest you take a look at the [Minimum Layer Time](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cooling/cool_min_layer_time.md). Increasing the minimum layer time will force Cura to slow down so that the layer has enough time to cool down. There are also other settings like fan speed (which should probably always be at 100% for PLA anyway).

sw35, 2020-08-20:
I use a single nozzle and will not have these problems 

I have this problem with 2-in-1 Hothend(mixed nozzle)

I think it should be the sprinkler, the material will drip out when the sprinkler is retracting, as shown below
[image]

It is true that lowering the nozzle temperature can improve these problems, but the strength and combination of the printed model will be very poor.

I use two different materials, such as PLA and PVA (for support), because of set a higher material temperature
Will make these problems more frequent

Sometimes, because of these problems, it may lead to model failure or model misalignment

That’s why I hope to have this function to print out liquefied materials in advance

Ghostkeeper, 2020-08-24:
So if you only use the first extruder in a print it's fine, but if you've used the second extruder for a few layers 12 hours ago, the first extruder will continue to print badly for the rest of the print? I really don't understand. After the second extruder has printed its blue layers, the print is effectively single-extruder from then on. The second extruder is retracted and gone.

It shouldn't need to heat up to the PVA printing temperature any more or anything. If you're printing PLA+PVA but the layers with PVA is done, it should be printing PLA at 190 degrees or something for the rest of the print. It's also printing PLA at 190 degrees if you're printing single extrusion. How could it ooze differently if at some point in archaeological history it switched materials?

sw35, 2020-08-24:
I'm sorry, I clarify the hotend I use
I use 2-in-1 Hothend, 2 extruders with shared heater (mixed nozzle).
As shown below
[image]

Both materials are heated together, only one temperature can be set
[image]

That's why there will be problems when the hotend retracts and moves.

I use this hotend because I need to mix colors in my work
But there is also a need for two-color, two-material applications.

Ghostkeeper, 2020-08-26:
Yeah so what is the problem with that hot-end then if there used to be a different material in the nozzle chamber at some point several days ago, that would require you to switch back and forth at every layer days later?

Ghostkeeper, 2020-08-26:
Maybe it just needs a longer Nozzle Switch Retraction Distance, to pull the inactive material out of the hot zone?

closed 2020-09-09 by no-response[bot]