Ultimaker/Cura #8167: Cura improvement suggestion. 'Crossline in prime tower'

thatgirl0126, 2020-08-10:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

Hi. Hope you are well.
As one of the Cura user, I sincerely appreciate the time and effort of your team involved in improving the free slicing software for everyone. I am using Cura 4.6.2, and I have some suggestions for the Cura improvement for better printing experience.
 
1. Crossline in prime tower
- Problem: In dual nozzle printing, sometimes, the 2nd nozzle extrudes the material in the air during prime tower printing. I found a reply says that the extruded material from the 2nd nozzle will be dropped into the basket, but in case the height of the basket is not enough (0.2~10 mm), 2nd nozzle movement makes tumbleweed by extruded material during prime tower printing, and it is dragged to the prints.
[image]

 
- Suggestion & effect: I would appreciate it if you add a crossline to the inside of the circle tower where (the layer) the 2nd nozzle doesn’t extrude. Then, I think the extruded material from 2nd extruder can be fixed on the crossline, and it will not be dragged to the prints.
[image]
[image]

 
2. Nozzle switch retraction prime speed
- Problem: In nozzle switch, PVA(in 2nd nozzle) is primed around 5 seconds as the 'nozzle switch retraction prime speed', but the rest of the retracted length is primed too fast. (When I set the nozzle switch retraction distance to 30mm and prime speed to 2mm/s, 10mm are extruded as the setting speed for around 5 second, and the rest[20mm] are extruded too fast.) It often causes filament grinding or filament bending in extruder.

- Suggestion & effect: I think it will be good if I can secure enough 'prime time' so that the retracted filament can be pushed to the end of the nozzle slowly. (It will be good if the 'Nozzle switch retraction extra prime speed' is also same with it)
 

3. Printing speed control for every initial layer of the bottoms.
- Problem: When I print PLA, I used to set the nozzle temperature at 190~195 degree Celsius with 50 mm/s printing speed for better outer wall and overhang quality. And in case I print PLA with PVA, I decrease the printing speed of the Top/Bottom layer for the better layer adhesion between PLA and PVA, and it increases the printing time too much.

- Suggestion & effect: I hope I can reduce the printing speed of 'every initial layer' of the bottoms which contact with the support. I think this can improve the layer adhesion by giving enough melting time.

Hope everything goes well for you.
and thank you for your endeavor again,

nallath, 2020-08-11:
The first problem can be fixed by enabling the Ooze shield setting.

Ghostkeeper, 2020-08-11:
1. If you enable "Wipe Inactive Nozzle on Prime Tower" it creates an extra movement that wipes off any excess droopage on the prime tower too. Is that solution not sufficient for your printer?
2. If using retractions, it'll move first to the retracted position (using the nozzle switch prime speed) until it has completed a travel move to position the nozzle on where it first needs to start printing (or priming in the case of the prime tower). Then it'll unretract using the normal retraction speed. I wouldn't expect the retraction/unretraction speed to cause problems. If it does, you probably need to reduce the retraction speed. Would you expect it to behave differently?
3. To a certain extent this can be controlled with the Overhanging Wall Angle/Speed and bridging settings. But indeed we are missing a "bottom surface" settings set. This is a duplicate of https://github.com/Ultimaker/Cura/issues/3938.

thatgirl0126, 2020-08-12:
Really thank you for the kind explanation.

> The first problem can be fixed by enabling the Ooze shield setting.

thatgirl0126, 2020-08-12:
> The first problem can be fixed by enabling the Ooze shield setting.

thatgirl0126, 2020-08-12:
> The first problem can be fixed by enabling the Ooze shield setting.

Thank you for the good suggestion. well...I don't use ooze shield for dual nozzle printing using PVA, because the ooze shield is often broken by filament lump (I described above) although I always set the shield angle to 0 degree for the better strength, and it causes problem...T.T I think it will be good if we can print ooze shield by only 1st nozzle material.

thatgirl0126, 2020-08-12:
> 1. If you enable "Wipe Inactive Nozzle on Prime Tower" it creates an extra movement that wipes off any excess droopage on the prime tower too. Is that solution not sufficient for your printer?
> 2. If [...]

Really thanks for the kind explanation.

1. I activated ‘Wipe Inactive Nozzle on Prime Tower’. The filament lump on 2nd nozzle is sometimes wiped by ‘Wipe Inactive Nozzle on Prime Tower’ function. But it (the filament lump) still exists inside of the prime tower, and because the height of the filament lump is much higher than tower wall (in case the wall height is 5 mm below), it adhere on the 1st or 2nd nozzle again, and it is dragged to the prints…^_^;;

2. I couldn’t think that the ‘retraction retract speed’ is related with the ‘nozzle switching prime speed’. I will test it. Thanks.

*PVA settings
-	Thinkness: 1.75 mm
-	Retraction distance: 6 mm
-	Retraction speed: 30 mm/s
-	Prime speed: 10 mm/s
-	Switch retraction distance: 25 mm
-	Switch retraction speed: 30 mm/s
-	Switch prime speed: 2~5 mm/s

3. It seems like I need to study more about the cura settings..^_^;;;. 
    Thanks for the guidance. I will refer it.

Ghostkeeper, 2020-08-13:
You get a 5mm tall blip?! That's huge. But then a crossing line to wipe it off also wouldn't be very effective, I'd think.

thatgirl0126, 2020-08-24:
Yes, it is because I use to extrude a lot of PVA filament during the prime tower printing to remove the expanded filament part. (extruding + rolling = huge lump!!) Well...so, I think the purpose of the crossing line is not a 'wiping'. I expected that it can be used as a 'support structure for 2nd nozzle material' in the prime tower. 

Additionally, I tested slower retraction speed for the 2nd nozzle as your suggestion.
In my case, it caused more strings....T.T

Thank you anyway!!^_^