Ultimaker/Cura #8171: [4.7.0-B] Is math Possible

FlailAway, 2020-08-10:
**Is your feature request related to a problem? Please describe.**
I'd like to use the Marlin look-ahead temp

**Describe the solution you'd like**
`Within the Start/End GCode?

;  M109 S210 B230 F1  ; autotemp PLA

M109 S{material_print_temperature}    **B{material_print_temperature}  + 20**    F1
OR
M109 S{material_print_temperature}    **B{material_print_temperature + 20}**    F1
`
**Describe alternatives you've considered**
None, too lazy

**Affected users and/or printers**
All/Any probably running Marlin

**Additional context**
Takes too many lines to make the green "Submit" button become active. ;)

fieldOfView, 2020-08-11:
No, it is not possible to use math in the start/end gcode snippets.

FlailAway, 2020-08-11:
Ummm, thanks, I know that. It is a feature request and pretty sure that would exclude existing features.

Ghostkeeper, 2020-08-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/4611.

closed 2020-08-11 by Ghostkeeper

fieldOfView, 2020-08-11:
> Ummm, thanks, I know that.

Hey, I am sorry; the subject of your issue - while missing a questionmark - sounds like a question. I answered that question.