Ultimaker/Cura #8174: Adaptive layers Min/Max Layer Hieght

360alaska, 2020-08-11:
I would like the option to set MIN/MAX. This may seem redundant, but it will give greater control than max variation.

Asterchades, 2020-08-11:
It gives the same control, just with different labels. For example, if you want a minimum of 0.12 and a maximum of 0.32, you presently set your layer height to 0.22 with a maximum variation of 0.10. Minimum of 0.8 and a maximum of 0.24 would be 0.16 layer height and 0.08 variation. layer height = (minimum + maximum) / 2, variation = (maximum - minimum) / 2.

The only real difference would be trying to determine whether your chosen overall layer height was the minimum, maximum, or ignored completely. It may also be slightly more read-able, but there's nothing to stop limits calculated with the above from being displayed (and I guess optionally being overridden, like you would the number of walls as compared to the wall width).

nallath, 2020-08-11:
The deviation is actually better, since the settings that you select are usually strongly based on the layerheight. 

So if you currently select a print with a layerheight of 0.1 and a deviation of 0.05, you get a print with layers between 0.05 and 0.15, but with settings as if it's sliced for 0.1

With the min/max, you could set it up to use a 0.05 profile, but with the max set to 0.15. You would use a profile that is less like what you would expect.

Ghostkeeper, 2020-08-12:
Agreed. Asterchades' explanation explains why it's unnecessary. [This duplicate issue](https://github.com/Ultimaker/Cura/issues/2895) argues that it's easier to conceptualise for the user, but Nallath's explanation explains why it's worse than the current situation. Using profile settings made for the average layer height is optimal and important! So I think we'll reject this feature request.

closed 2020-08-12 by Ghostkeeper