Ultimaker/Cura #8175: Slicing generates broken tree supports

RobVilchis, 2020-08-11:
**Application version**
4.6.2

**Platform**
Windows 10

**Printer**
Creality Ender 3 and CR-10

**Reproduction steps**
1. Import the project attached below
2. Slice
3. Analize print preview
4. The issue starts aproximately at layer 65

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
At a certain height, the supports move a little to the side, and start being printed in mid-air. This has caused a lot of failed prints.

**Expected results**
Functioning tree supports

**Project file**
[CCR10_M0119588.zip](https://github.com/Ultimaker/Cura/files/5054237/CCR10_M0119588.zip)

nallath, 2020-08-11:
Could you try this again with the 4.7 beta? A number of fixes have been made for it.

Ghostkeeper, 2020-08-12:
You are really reporting two bugs. One of them, with the small piece of support starting in mid-air, is still open, but a duplicate of https://github.com/Ultimaker/Cura/issues/4342 .

The other bug, with the lines being placed beside each other instead of on top of each other, could be one of two things:
* Tree Support Collision Resolution is too high (too low resolution). This makes the collision volume inaccurate and can cause big jumps in the position of the tree support branches. I would advise to set that to 1/2 line width at most.
* When merging branches of different radius together, it would sometimes have a jump in the branch radius. This has been fixed for 4.7 so if this is the issue it should be gone in 4.7.

And indeed the second problem doesn't seem to reproduce in 4.7 any more, when loading that project file. So this bug seems to already have been fixed!

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/4342 then.

closed 2020-08-12 by Ghostkeeper