Ultimaker/Cura #8176: [4.7.0-beta] No extruder temperature in 'Monitor' session.

protywu, 2020-08-11:
**Application version**
4.7.0-beta

**Platform**
Win 10, 64 bit.

**Printer**
Several including Ultimaker S5.

**Reproduction steps**
Simply open the software and click on the 'Monitor' tab.
Also tried running Cura after powering on the printer so that the printer was automatically connected to Cura when Cura is up.

**Screenshot(s)**
[image]

**Actual results**
The 'extruder temperature' session is missing.

**Expected results**
The 'extruder temperature' session should show up as in the versions before 4.7.0-beta.

**Project file**
N/A

**Log file**
N/A

**Additional information**
Have tried to manually put in the MKS Cura plugin into the 'plugins' directory and the results were the same, no 'extruder temperature'.

nallath, 2020-08-11:
Please contact the developer of the MKS Cura Plugin. We can't control what plugins do / show.

closed 2020-08-11 by nallath

protywu, 2020-08-11:
> Please contact the developer of the MKS Cura Plugin. We can't control what plugins do / show.

The plugin comes with the Cura official download, so we don't know where the official Cura got the plugin from.

nallath, 2020-08-11:
The MKS plugin is not bundled with the Ultimaker Cura release. It is available on the market place, where you can also find the contact information of the developer (https://marketplace.ultimaker.com/app/cura/plugins/Jeredian/MKSWifiPlugin).