Ultimaker/Cura #8177: Refresh network connection to printer

olubeck, 2020-08-11:
I frequently have to refresh the network connection to my printer to send a file.
It's a very annoying UI task that must be done to enable the "Send Over Network" button.

Why not enable the "Send Over Network" button all the time if printer was setup and refresh automatically rather than require the current UI.

I can do exactly this on all my 2d printers. I simply say print and the print driver refreshes the connection to the printer. 

Olaf Lubeck

nallath, 2020-08-11:
That's exactly what it's supposed to do. So could you please fill in the bug template? 

Also; are you using the Octoprint plugin or are you sending it to an Ultimaker?

olubeck, 2020-08-11:
I am sending it to an Ultimaker directly.
I have 2 -  an S3 and S5.
At different locations.

S5 is connected via wifi, the S3 is connected via hardwire to a router.
Sometimes Cura recognizes that there is a printer on the network and enables the “Print Over Network”, but more often it doesn’t.

In fact, as I’m writing this, the button is toggling back and forth from “Save to File” and “Print Over Network”.
This is on my S3 which is hardwired to my router.

Olaf

nallath, 2020-08-11:
Right, have you tried adding the printer directly via IP? Some networks have issues with the autodetect broadcasting.

Alternatively, you could also connect them via cloud, then you also don't have this issue.

olubeck, 2020-08-11:
Thanks. I”ll try these solutions.

In the connection model that I’m envisioning, I don’t see why the button toggles at all.
If the user setup a network connection, then the “Print Over Network” button should be enabled no matter what the current connection status is.
Maybe need a preference item to allow “Print To File” even though user has setup a networked printer.

Olaf

fieldOfView, 2020-08-11:
I recently wrote a small plugin for someone with the same issue. It is a bit rough, and not ready for primetime, but it should let you work around your issue. It comes with the following caveats:

* The button does not show up unless there is a model on the buildplate.
* The button shows up regardless of what printer is selected
* The buttons has no visible feedback
* The plugin makes me cry (because both implementation-wise and UX-wise it is a hack)

To install this plugin, download the [plugin package](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/NetworkRefreshButton/nkOcs-ysJI78zwBPt9l29g3LJfG2Ih2HPdPz8YCHNZ5z/NetworkRefreshButton-v7.0.0-2020-07-19T20%3A17%3A19Z.curapackage?Expires=1912518871&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=C0HHa8M553JsyttXzS5MWcnsVGvTPSOJ6cws4KqDvqmMGLyDGZcMDrT2nzI%2F2L1QafDpy3Ib%2B2%2FRT0wsHseNGKU81vLCKhaQHyoYXQUJreS6b35JnITlHya5%2F5xf3p9Qz5LISZQ9JKvJM2bFGxJeCwv81Ozrjw%2BHjcF2lmZWOI%2BU%2FNR4Zcs%2F4sqVRC5ioxml3eP74E22UYpk8CHzQWE%2Fw%2BrNRU%2BkFcDmk1fcG5jdEi5gV5vKrM3bYKt8Ri7bKkMWQ%2FKij5RWFb1f4i4H9VWYj3tUNBOeqhoNfuHZIaRM0jqcgFRo2j9oLZJRUwyVxkfpuVKHEbHhZWzA92AQbmeJHQ%3D%3D) and drop the file onto the buildplate in Cura, as if you were opening a 3d model. Restart Cura, and a "refresh" button should be added to the slice/save/print area, with the above limitations.

nallath, 2020-08-12:
> If the user setup a network connection, then the “Print Over Network” button should be enabled no matter what the current connection status is.

How would that make it any better? It hides the button because it can't reach the printer. So if the button would not be hidden, it would just kick the can down the road (because you're effectively in the same situation; it doesn't work)

Ghostkeeper, 2020-08-12:
I think he meant that Cura would save the print locally in a queue (possibly also showing it in the Monitor page), to be sent to the printer when the connection is restored.

Ghostkeeper, 2020-08-12:
> Maybe need a preference item to allow “Print To File” even though user has setup a networked printer.

Actually it is still possible to print to a file rather than print over network, even if you are currently connected to a printer. Just click on the little arrow on the side of the blue print button to select where you'd like to print to. In Cura 4.7 it'll also remember your last choice, here.

I've also seen this issue myself on my LAN. Restarting the printer (while leaving Cura open) also fixes the problem; then the printer announces itself on the network which gets picked up by Cura and reinstates the connection.

Printer IPs can change, but don't change very often usually. Maybe it could be a solution to store the printer's IP as a fallback in Cura, alongside with the network key? It should use the network key first, but if that fails to connect it can connect via IP. It's going to behave a bit weirdly with extensibility that way though, so maybe we'd need to implement that fallback only in the UM3NetworkConnection plug-in.

olubeck, 2020-08-12:
Below is a picture of the steps that the current UI requires:

1. User sees “Save to File” button, but wants to “Print over Network”.

2.So User selects Settings>Printer>Manage Printer 
3. User sees empty list of Networked Printers
4. User does nothing except select “Refresh” button.
5. “Print over Network” button is enabled.

Steps 2-5 happen often, but are annoying and unnecessary (I think).

Proposed Model:
A. User sees and selects “Print over Network” button. It’s always enabled (after printer setup).

B. Cura checks network status.
	B1. If bad - Cura executes processes in steps 2-5 above without UI. If still bad, then execute a warning UI (to turn on printer, etc).
	B2. If good - send file.

This seems more efficient.

Olaf

olubeck, 2020-08-12:
Thanks. Didn’t realize this.

Olaf

fieldOfView, 2020-08-12:
Have you tried the plugin I mentioned above? It is not ideal, but at least one user confirmed it worked to "rediscover" a lost connection with a network printer

olubeck, 2020-08-12:
No. Not yet.

I assume that I have to select it and go through some UI to refresh printer connection?

Olaf

olubeck, 2020-08-12:
Yes. I see.
Shortcuts the multiple steps of the current UI.

But no warning if, for example, the printer is turned off.
And don’t see need for a separate refresh button if refresh process is initiated by the “Print Over Network” button?

Olaf

nallath, 2020-08-13:
@olubeck; pictures don't work in the emails you send to github.

olubeck, 2020-08-13:
[CuraNetwork.pdf](https://github.com/Ultimaker/Cura/files/5069948/CuraNetwork.pdf)
Hope this supplies pictures