Ultimaker/Cura #8178: Slicing is "stuck". A Part that normally slices quickly gets stuck for hours slicing. Never completes.

gdaily19, 2020-08-11:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-08-11:
Without logs, and the other details that are requested in the issue template that you seem to have missed, there is preciously little that we can do to help you, unfortunately. So please fill in the issue template, and supply the requested logs.

gailulun, 2020-08-12:
Is your model file too big?

Ghostkeeper, 2020-08-14:
There is a limit of 512MB due to the message size limit in Protobuf. But if that happens the slice doesn't even start so it wouldn't appear as if the slicing is stuck.

closed 2020-08-26 by no-response[bot]