Ultimaker/Cura #8179: [4.6.2] Cannot remove local printer

FastThenLeft, 2020-08-11:
**Application version**
4.6.2

**Platform**
Windows and Ubuntu 20.04

**Printer**
I had previously created "Artillery Sidewinder X1 V4".

**Reproduction steps**
I'm trying to delete the local printer.
1. Go to Settings -> Printer -> Manage Printers...
2. Select the local printer I want to delete. In my case it's "Artillery Sidewinder X1 V4"
3. The "- Remove" button remains grayed out and I can't delete the printer.

**Actual results**
I can't delete the local printer.

**Expected results**
I should be able to delete the local printer.

rburema, 2020-08-12:
You can't delete a printer that's active. There always has to be at least one printer. If you switch to another printer, you should be able to delete the now non-active one.

closed 2020-08-12 by nallath

FastThenLeft, 2020-08-12:
Thanks! That fixed it. Sorry it wasn't obvious to me.
-David

rburema, 2020-08-12:
No need to apologise!

Ghostkeeper, 2020-08-13:
You can delete the active printer, but not the last one.