Ultimaker/Cura #8182: Cura 4.6.2 changing Z scale automatically changes Y scale after rotating to face (even with uniform scaling off)

eatinginternet, 2020-08-12:
**Application version**
Cura 4.6.2

**Platform**
HP Envy x360, Ryzen 2700U, 8GB RAM, Windows 10 x64
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Load model
2. Select face to rotate to build, and click on a face of the model
3. Unselect uniform scaling
4. Change scale of Y axis, then change scale of Z axis (both by typing numbers in text box)

**Screenshot(s)**
[image]
[image]

**Actual results**
Every time you now change the y axis, the z axis changes as well and vice versa. Even just clicking on a scale textbox and clicking out will cause one of the two to randomly change

**Expected results**
The Y axis and Z axis should change independently even after rotating to a certain face. They should not change on their own

**Project file**
[CE3_clock v10.zip](https://github.com/Ultimaker/Cura/files/5065923/CE3_clock.v10.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Have fun

rburema, 2020-08-13:
Probably yet another duplicate of #4136 , a longstanding issue that proved surprisingly tenacious.

This is a new variant though, we should keep the 'rotate face to BP' route for this bug in mind while fixing the other one!

eatinginternet, 2020-08-13:
My bad, I'm new to Git. I searched but I figured bugs from that long were no longer being worked on / an issue so I didn't pay attention to anything older than the last few versions when I searched for duplicates.

rburema, 2020-08-13:
We tend to close duplicates as well, so discussion can take place in the original one.

This one is a very longstanding one, so I can see why it lead you to the conclusion that this hadn't been discussed yet!

closed 2020-08-13 by rburema