Ultimaker/Cura #8183: No nozzle size in cura with artillery genius profile

nflyer, 2020-08-12:
When i import a profile for artillery genius i can't see nozzle size ( in the nozzle  screen i see "empty")
-->

**Application version**
Cura 4.6.1

**Platform**
Windows 10 intel I7

**Printer**
Import profile for artillery genius

[image]

rburema, 2020-08-13:
Can you post the profile here?

(Anything in the logs when you imported it?)

theodore0310, 2020-08-13:
I have the same problem, with a pro

[petg myself.zip](https://github.com/Ultimaker/Cura/files/5068384/petg.myself.zip)

file for the artillery sidewinder x1, exported from Cura slicer of my old PC. This is it. ###

nallath, 2020-08-13:
It's not really something we can test, because we don't _have_ the artillery sidewinder machine definition. Where did you get that from?

nflyer, 2020-08-13:
from 3d nexus site

Ghostkeeper, 2020-08-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/8104

The profile on 3DNexus is simply outdated. Use Cura 4.6.1 or manually update the profile from 3DNexus.

closed 2020-08-14 by Ghostkeeper

PocoComunicativo, 2020-10-01:
I had the same problem. I have Ultimate Cura version 4.7.1 and I just download this profile, and works just fine:

https://www.thingiverse.com/thing:4581098