Ultimaker/Cura #8184: BUG: Different material selection does not change bed temperature - Cura 4.6.2

reefwarrior, 2020-08-13:
Hi all,

**Application version**
Cura Slicer 4.6.2

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Reproduction steps**
1. Define material with print temperature of 197, bed temperature 62)
2. Define material with print temperature of 235, bed temperature 82)
3. Select either; and check in print settings. Print temperature changes, but bed temperature does not. 

**Screenshot(s)**
[image]
[image]

**Actual results**
As discussed under 3.

**Expected results**
Bed temperature needs to update.

Rgds
Tinus

Ghostkeeper, 2020-08-14:
I'm not able to reproduce this problem. It updates fine for me:
[image]
[image]

Can you provide us with a project file? Like the template asked for, before you deleted that?

reefwarrior, 2020-08-14:
Hello,

Sure, included please find. I have also included the generated .gcode files, if that helps.

Thank you for the help!
Tinus

PS. Included in the .zip file
[CURA Issue.zip](https://github.com/Ultimaker/Cura/files/5073914/CURA.Issue.zip)

nallath, 2020-08-17:
That's because your custom quality profile overrides the material temperature. I've fixed that for you.

For the next time, you can see that this is going on by looking at the text of the setting. If it's italic, your profile is overriding it. 
[fixedprofile.zip](https://github.com/Ultimaker/Cura/files/5083400/fixedprofile.zip)

closed 2020-08-17 by nallath