Ultimaker/Cura #8185: FR - Different Z offset depending on material

reefwarrior, 2020-08-13:
Hi,

**Is your feature request related to a problem? Please describe.**
No. I need a different Z offset to be specifiable depending upon material. Eg the Z-offset for my PETG must be 0.05 LESS than for my PLA as example.

**Describe the solution you'd like**
A way to automatically retrieve the current Z offset (M851); and then apply the 'delta Z' needed for the correct material to be printed/sliced. This is then added to the beginning of the G-code, eg. Selecting PETG will cause M851 to be executed; it gets eg. Z-1.00 - in the materials definition it was specified as DELTA Z to be -0.05. It issues M851 Z-1.05 in the beginning after the G28 in the startup G-code.

**Describe alternatives you've considered**
Doing it manually, but it is quite time consuming.

**Affected users and/or printers**
Users that print different materials, requiring a different offset with regards to the material being printed.

Rgds
Tinus

fieldOfView, 2020-08-13:
Use the Z Offset plugin in combination with Material Settings plugin.

Ghostkeeper, 2020-08-14:
I would also recommend playing with the Initial Layer Height and Initial Layer Flow first; it should achieve a better vertical dimensional accuracy then.

Expanding the list of settings available in the material menu is still something I'm keeping in the back of the mind since it's quite a common request. But it's hard to make material profiles properly understandable to the user how the profiles can override each other, and fieldOfView's plug-in does as good a job as any.

reefwarrior, 2020-08-14:
Hi! Thank you! Will play around with it for the time being. Shall I close this ticket and 're-open' if not able to get this working?

Ghostkeeper, 2020-08-17:
In that case you might run into a bug with those plug-ins. It should be possible.

mrx23dot, 2020-12-19:
+1 for Z-offset as part of core features, there is no point installing an extra plugin for such a small feature.
And it changes with the material profile.

closed 2020-12-21 by nallath

mrx23dot, 2020-12-21:
Is this a "won't fix" ?

nallath, 2020-12-21:
It's a "Whoops why is this still open since there is a solution for it".