Ultimaker/Cura #8186: Decouple printer definition from network printer name

BagelOrb, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
A colleague has been working on a project on his Ultimaker S5 called ColleaguePrinter and sends me the file. I have the same setup at home, but a different Ultimaker S5 called PhdPrinter.
Now it's impossible for me to network print his project on my printer.
The setting changes are saved on ColleaguePrinter, but when I switch to PhdPrinter all the setting changes will be gone.
If I go to manage printers and connect ColleaguePrinter via network to PhdPrinter, then the ColleagePrinter is renamed to PhdPrinter, even though I already have a printer definition called PhdPrinter.
This means that there is now 2 printers called PhdPrinter - one with my settings and one with the settings from my colleage.
I have no way of knowing which is which.
Also it is not possible to rename the printer once it is connected, so Cura actively makes it impossible to distinguish the two.

**Describe the solution you'd like**
The printer should be named by the Cura user. The network name could be added in brackets.
e.g.
[image]

**Describe alternatives you've considered**
- Only mentioning the network name somewhere else in the Manage Printers screen or the Connect via Network screen.
- Making the network name the main name and the user given name in brackets.
- Only mention the network name in the Print button, i.e. [Print over Network to PhdPrinter]

**Affected users and/or printers**
Users of Ultimaker 3 and later printers.

**Additional context**
When working on a project it is important to be sure I am printing with the intended settings, but currently I can only be sure by deleting my own PhdPrinter and throwing all my settings away before reconnecting ColleagePrinter to PhdPrinter.

konskarm, 2020-08-13:
We have added a new feature where you will be allowed to select which existing cura printer to update when you open a project file. The printer you select to update will not have its network information altered, so it will basically remain the same. All the other project information will be properly loaded. This feature will be available in 4.8: https://github.com/Ultimaker/Cura/pull/8144. If you want, give it a try in the latest master.

BagelOrb, 2020-08-13:
Hmm. I guess that solves my original problem, but now I still have two printers with the same name which I cannot rename.

BagelOrb, 2020-08-13:
It's not a really big issue, it's just rather annoying that you can't rename a printer definition to what you want.

konskarm, 2020-08-13:
We noticed that issue too. It seems that we can have up to three printers with the same name, a local, an IP-networked, and a cloud printer. We already have a ticket in our backlog for that (devs see CURA-7652). 
Nevertheless, I'll put your current issue up for discussion, maybe we can show at least the way the printer is connected next to its name (e.g. Big-momma (cloud)) or something similar.

BagelOrb, 2020-08-13:
I have the same printer as many times as I want in Cura, all via IP. I don't see that as an issue. That's good.

But I don't really understand that you have made it impossible to rename the printer. With 2D printer you also always have the choice yourself what to call it on your computer. I don't understand that Cura fixes the name to what it's called on the network.

Of course it's nice that it defaults to that name, but in the end I think it should be up to the user of Cura to call the printer what they like - it shouldn't be up to the person managing the printer what I call that printer on my computer.

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.