Ultimaker/Cura #8187: Layer view too black

BagelOrb, 2020-08-13:
**Application version**
Current master

**Platform**
Ubuntu

**Printer**
S5

**Reproduction steps**
1. Use PC black
2. Slice
3. Go to layer view

**Screenshot(s)**
[image]

**Actual results**
Can't see extrusion lines
Color is too black

**Expected results**
Color should be dark grey, so that I can see the separate layer lines.

**Project file**

**Log file**

**Additional information**
Might wanna check other black materials for being too dark as well.

Ghostkeeper, 2020-08-14:
We're doing transformations on the colour here to make it more visible. As a result, the `#0e0e10` colour of this material ends up at `#292933` in the actual layer view. (In decimal: `[14, 14, 16]` to `[41, 41, 44]`.)

Whether or not this is visible depends a lot on the screen I guess. I'd expect the difference between #29 and #00 to be pretty clearly visible on most screens. We can try cranking it up but it's a bit of a guess whether it will be enough on other people's screens, and you need to be careful to not wash out the difference between materials at high luminosities.

fieldOfView, 2020-08-14:
You could experiment with adjusting the amount of shininess for darker colors.

Ghostkeeper, 2021-04-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/5264.

closed 2021-04-03 by Ghostkeeper