Ultimaker/Cura #8188: [FR] Remove Unmerged Support When Support Interface Is Present

Block137, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
There's an undesirable (to me) behavior described at https://github.com/Ultimaker/Cura/issues/7157 which massively increase print time but does not give benefit under certain condition.
In simple term: when support layers are merged (support printed in lower Z resolution), there are fragments of unmerged support which will be printed separately. These fragments were intentionally kept to improve adhesion with model but results in insane amount of retraction between every fragment.

**Describe the solution you'd like**
Remove unmerged support fragments when support interface or support wall is enabled. The fragments no longer support the model in these cases.

**Describe alternatives you've considered**
Create a(n external) post processing script to filter out islands with small extrusion amount. The script also check for ;TYPE: so it is a little more versatile than Minimum polygon circumference. This create an unnecessary extra step.

**Affected users and/or printers**
User who set support layer height to multiple.

**Additional context**
Description, image, example, project file can be found here:
https://github.com/Ultimaker/Cura/issues/7157

Ghostkeeper, 2020-08-17:
Not sure how beneficial this really is. The same extra layers are responsible for filling the gap if the support interface starts at a height that is not divisible by the Support Infill Layer Thickness setting. Like in this case, which I made a bit more extreme to make it visible in layer view:
[image]

Leaving out the support there gives a void in the support, which can cause the support to fail, especially if there's a lot of material resting on top of it.

Block137, 2020-08-17:
It's a bit more complicated than I thought.
The problem is not the extra layers but the way they are fragmented.
One solution I can think of is to extend support interface downward to reach support infill and replace extra layers/fragments at the same time.

I've made a script which removes all support path that extrude less than 0.5mm^3 between retracts. It was able to cut the print time down by 11%.