Ultimaker/Cura #8190: [4.6.1] Print Profiles for Custom Materials and Colors

derwon90, 2020-08-14:
**Application version**
4.6.1

**Platform**
Windows

**Printer**
S5

**Reproduction steps**
1. Created custom generic PLA material with a color (RED)
2. Created custom print profile for generic PLA RED
3. Open cura and have model set to Generic PLA RED

**Screenshot(s)**
1.	Open up cura, load model and have Generic PLA RED set for material – no custom print profiles available
[image]

2.	Change material to Generic PLA – all print profiles available, including the custom profiles
[image]

3.	Change material back to Generic PLA RED – now my custom print profiles are available
[image]

**Actual results**
1. No Custom Print profile shows up

**Expected results**
1. The custom print profiles should show up in the drop down menu.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
We use the S5 with the material station we're trying to streamline a way to use generic materials in the material station so that we can program prints with specific colors of PLA, to do this, we would need to create new generic PLA materials and have the colors set for each one. This would be a nice feature request.

nallath, 2020-08-14:
How did you create the new material? I've not been able to reproduce it. 

My steps were:
1. Duplicate generic PLA
2. Change the color of the new material
3. Press the "unlink" button (since that creates a new GUID, which is what you'd want for this)
4. Activate the red material

I still have all profiles available / selectable.

derwon90, 2020-08-14:
I think I've realized my issue. When I duplicate the material, I also change the material type to "PLA *COLOR*" and that disables the rest of the print profiles.
[image]

This still won't allow me to set various colors into the material station. When I first started playing with this, I realized that the only way "PLA RED" will show up as a material in the S5 printer is if I create a new material type "PLA RED". Even if I create a new profile with the following:
Display name: PLA RED
Brand: Generic
Material Type: PLA
Color: Red

The machine doesn't separate each spool by color but only by the Material Type so any generic PLA spool I load into the station will only be known as a generic PLA, regardless of color. There is no way I can set color to a generic PLA material loaded into the machine.

nallath, 2020-08-17:
I'm not a s5 firmware expert (and since everyone is working from home, I can't test this for you on an s5), but what happens if you just change the brand? You could just set it to "Custom". In that case, I think it would identify it as another material. 

Also, take care to ensure that the GUID of the material file is not the same.

Ghostkeeper, 2020-08-17:
This is something that Ultimaker's material partners (submitting profiles to the Marketplace) have also reported. Quality Assurance (Mari) explained this to me as the reason why those partners keep creating new material types rather than to just say that they are PLA.

Blame is put with the Opinicus firmware. It seems to be done to prevent the material selection menu from being too long due to all of the colours of Ultimaker brand materials available. They group all materials of the same material type together under one denominator (and select the Generic profile then when you click that option, if available). Instead they should group all materials that have various colours but are the same for the other properties (material type, brand and name).
As far as I know this has not been escalated (yet) though.