Ultimaker/Cura #8191: Color Options for Generic/3rd Party Materials

derwon90, 2020-08-14:
**Is your feature request related to a problem? Please describe.**
Print Profiles for Custom Material Colors
Issue 8190, https://github.com/Ultimaker/Cura/issues/8190

**Describe the solution you'd like**
We use the S5 with the material handling station, the printer is equipped with NFC reader to know what NFC enabled filament is being loaded as well as the color. there is no way to set the colors for generic materials loaded into the printer. It would be nice to have the option to toggle the color on the material screen for generic materials loaded into the printer. Cura would also need to have the ability to program parts in various colors of generic material.

**Describe alternatives you've considered**
Current alternative we're using is to just create new materials with the color set for it, like "Generic PLA RED" and "Generic PLA BLACK". the issue is that the print profiles sometimes has a bug and the other print intent profiles from PLA do not transfer over, like Draft and Engineering

**Affected users and/or printers**
All users who print multi color parts with generic/3rd party filament.

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-08-14:
If you Duplicate the Generic PLA material in Cura, you can change the color and name it Red PLA. You can use this Red PLA in Cura, and the S5 (and I think the material station too) should accept it as the Generic material.

nallath, 2020-08-14:
If you select an Ultimaker material with a color, it will just use the same settings as generic. So in that sense, you can just "trick" the material station into using colored materials.

Ghostkeeper, 2020-08-17:
Sort of the point of the generic materials is the ability to choose something that's good enough but not specific to a certain actual filament, colour or brand.

The generic materials are not communicated to the printer, since it already has these materials. It's not possible to have Cura change the colour of the material on the printer firmware.

closed 2020-08-17 by Ghostkeeper