Ultimaker/Cura #8192: 4.7 beta "Connect Support Lines" option only available when "Grid" is selected

GregValiant, 2020-08-14:
**4.7 beta**
(The version of the application this issue occurs with.)

**Win 10 Home 64bit**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Creality Ender 3 Pro**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Select - Support/Support Pattern/Lines the "Connect Support Lines" option gets hidden
2. Select Support/Support Pattern/Grid the "Connect Support Lines" option is available

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Select the Grid option and then the Connect Support Lines option becomes available.  Enable Connect Support Lines and then switch the Grid option back to Lines.  The lines do get connected.

**Expected results**
Whether selecting either Grid or Lines, the Connect Support Lines option should be available.
(When selecting ZigZag the Connect Support ZigZags option is correctly displayed.)

**Project file**
Not applicable

**Log file**
Not applicable

**Additional information**
Not Applicable

rburema, 2020-08-16:
ZigZag support is basically equivalent to what Lines + 'Connect Support Lines' would be. For completions sake we could consider adding 'Lines' to the 'show if' ('enable') field of 'Connect Support Lines'.

GregValiant, 2020-08-16:
[image]
[image]
[image]

In the first screenshot the Support Pattern "Grid" is selected.  The "Connect Support Lines" option is visible and selectable.
In the second screenshot the Support Pattern "Lines" is selected.  There is no option to "Connect Support Lines".  I believe it should be visible.  The work-around is to select "Grid", select Connect Support Lines, and then switch back to "Lines".
In the Third screenshot the Support Pattern "ZigZag" is selected.  The "Connect Support ZigZags" option is visible and selectable.

Ghostkeeper, 2020-08-17:
Yeah, this is an inconsistency, for two reasons:
* Settings that are not visible in the interface should not influence the slice result (aside from explicitly invisible settings such as the machine settings).
* In the infill settings, the Connect Infill Lines setting is visible even when the infill pattern is "Lines". It effectively then turns the infill pattern into zigzag except it doesn't connect the endpoints.

Note that "Connect Support ZigZags" is an entirely separate feature specific to the zigzag pattern. It's got little to do with this bug.

So I guess this is easy enough to fix to solve both inconsistencies at once: Just make the setting visible for the lines pattern in support as well.

closed 2020-08-17 by Ghostkeeper

GregValiant, 2020-08-17:
Works for me.  Thank You.

commit 6d90c85f (2020-08-17): Also show Connect Support Lines for lines pattern

Previously this was disabled because you should probably use the ZigZag pattern then. But I'm enabling it now because of two inconsistencies:
* The setting still works. So you can change the setting by switching to grid, triangles, cross or gyroid, then switch back to lines and still get connected support lines.
* The same happens with the infill patterns, where this setting is visible for the lines pattern.

Rather than changing the other two things (which involves changes to CuraEngine too to make it ignore the setting value for the lines pa [... cut]