Ultimaker/Cura #8193: GCode - Customize "Abort Print" GCode?

prometx, 2020-08-14:
**Application version**

4.6.2

**Platform**

 linux (ubuntu 18.04.2_amd64)

**Printer**

anet A8 (albeit heavily modified)

**Reproduction steps**

1. abort print
2. (Something you did next.)

**Screenshot(s)**

"N/A" 

**Actual results**

for my  particular machine, i find that "end gcode" for a print helps keep my hotend unclogged if i home the hotend after printing and "purge" the extruder and hotend of several centimeters of filament before letting the head cool down.

i find that on executing an "abort print", though that cura doesn't honor any of this end code, which leaves me scrambling on abort to manually prevent clogs.

i wonder if there is not some way to customize the "abort print gcode" to help make it fall inline with my custom "print end code", or maybe make it 'fail' to my end code, instead of its "stock" abort routine? which would be great and save me a lot of time and energy.

**Expected results**

not "expected" really, but that if "end code" is changed, "abort code" conforms the this as a "sane abort"

**Project file**

N/A

**Log file**

i am embarrassed to say that i dont know where this is...

**Additional information**

i feel like the above will be sufficient to "impugn my character"

best,

rburema, 2020-08-16:
I'm sorry, but there'll be a couple of problems with this that aren't easily solveable from Cura's side of things.

For every printer we just send gcode (or something that's a wrapper around that). The gcode we send is just a list of commands, without any conditionals or interrupts. As such, the end gcode is only that; the gcode at the end. It can't be 'jumped to' when the print is aborted.

As it stands, the abortion of a print is something that is handled by the printer, not Cura.

ChipCE, 2020-08-17:
I think in Marlin, there is a config for that <code>EVENT_GCODE_SD_STOP</code>

prometx, 2020-08-17:
> I think in Marlin, there is a config for that `EVENT_GCODE_SD_STOP`

Hey ChipCE, thanks very much! I will def have a look at that.

Ghostkeeper, 2020-08-17:
When printing a g-code file it is truly impossible for Cura to implement this, for the reasons rburema laid out. When printing via USB cable, it is theoretically possible. That would be a duplicate of https://github.com/Ultimaker/Cura/issues/7562.

closed 2020-08-17 by Ghostkeeper