Ultimaker/Cura #8198: Minimum layer time has no effect in Initial layer speed

carloshespanhol, 2020-08-15:
**Application version**
4.6 and 4.7

**Platform**
Linux

**Printer**
Custom

**Reproduction steps**
1. load small 10mm ring
2. Set Minimum Layer Time to 10s
3. Set Initial Layer Speed to 20mm/s
4. Set Minimum Speed to 10 mm/s

**Actual results**
First layer is faster than other layers

**Expected results**
I would expect first layer speed to be limited by Minimum Layer Time

smartavionics, 2020-08-15:
Please provide a project file. Do File->Save and then either rename or zip the .3mf file and attach to this issue. Thanks.

Ghostkeeper, 2020-08-17:
Yeah, I'm unable to reproduce this issue. Here is a project file created with 4.7 that changes those settings. I used a cube rather than a ring, and set the adhesion also to None to ensure the initial layer printed in less than 10 seconds.
[minimum_layer_time_initial_layer.zip](https://github.com/Ultimaker/Cura/files/5085128/minimum_layer_time_initial_layer.zip)

This results in the following relevant snippet of g-code:

```
;LAYER:0
M107
;MESH:cube.stl
G0 F2400 X50.203 Y50.203 Z0.3
;TYPE:WALL-OUTER
G1 F1500 E0
G1 F600 X49.797 Y50.203 E0.00764        ; <--- Prints at 600mm/min
G1 X49.797 Y49.797 E0.01527
```

So it starts printing at 600mm/min, which is 10mm/s. That's the minimum speed, not the initial layer speed, so the initial layer time has been applied properly.

carloshespanhol, 2020-08-18:
You are right, 
what was causing the first layer to be faster was the build plate adhesion.
It is correct.

closed 2020-08-18 by carloshespanhol