Ultimaker/Cura #8199: [4.6.2] pause at layer Repetier

SBWZFH, 2020-08-15:
**Application version**
4.6.2
4.6.1
4.6.0
(maybe even older)

**Platform**
WINDOWS 10   1050gtx

**Printer**
Custom

**Reproduction steps**
1. Select pause at layer height for 3 separate layers
2. Confirm GCODE

**Actual results**
All instances of "Pause" is applied only to the earliest layer. Also, Z height returns 1mm above desired height. 

**Expected results**
Pause at layer should happen on the different specified layers. 

**Project file**  See end of layer 8 begin of layer 9
[CFFFP_Snorkle to fender grommettest.zip](https://github.com/Ultimaker/Cura/files/5079621/CFFFP_Snorkle.to.fender.grommettest.zip)

rburema, 2020-08-16:
I _think_ this might be fixed in 4.7-beta, could you try that?

SBWZFH, 2020-08-16:
> WARNING: This message has originated from an External Source. This may be a phishing expedition that can result in unauthorized access to our IT System. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Downloading now.  Thanks!

Stephen Williams
University of Missouri Science and Technology
Undergraduate in Mechanical Engineering
Cell: (816) 645-6715
Email: SBWZFH@umsystem.edu

closed 2020-08-31 by no-response[bot]