Ultimaker/Cura #8201: A way to set custom GCODE or linear advancke K based on filament type

ChipCE, 2020-08-16:
**Is your feature request related to a problem? Please describe.**
Each filament have their K value for linear advance so it's would be nice to have a way to set the K value for each filament.
For example: PLA 0.04 , PETG 0.046

**Describe the solution you'd like**
Linear advance setting (or custom GCODE) for each filament.

**Describe alternatives you've considered**
I manually edit the exported gcode file and insert the value, but its time consuming and error prone.

smartavionics, 2020-08-16:
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/LinearAdvanceSettingPlugin

ChipCE, 2020-08-17:
> https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/LinearAdvanceSettingPlugin

The plugin is for global setting(not per filament) and won't be exported with export function. I mean filament should have their own setting.

smartavionics, 2020-08-17:
Understood.

nallath, 2020-08-17:
I think that if you also install the materialsettings plugin that you can add the setting to the material file.

ChipCE, 2020-08-17:
@nallath  I think that plugin will only show/hide setting based on filament type.

Ghostkeeper, 2020-08-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/8156, from 11 days ago.

closed 2020-08-17 by Ghostkeeper