Ultimaker/Cura #8202: [4.6.2] Flow Rate Compensation breaks retraction priming.

ClockeNessMnstr, 2020-08-16:
**Application version**
4.6.2

**Platform**
Windows

**Printer**
Ender 3 V2. 

**Reproduction steps**
1. Prepare accelerationtower.stl from Teaching Tech 
https://teachingtechyt.github.io/calibration.html#accel

2. slice with:
  retract at layer change. (gcode has extra prime 0.064mm^3)
  filter out small gaps. 
  wall speed 112.5 mm/s
  accel 800mm/s^2
  4mm retraction 
  30 mm/s retraction speed

3. print to see failure on starting line of each layer.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**

 when printing, the first line of the layer change had been retracted 4mm + last commented flow rate compensation
 priming only primes 4mm. 
this leads to a gap followed by a lump of material approx 1/2 way through the long line move. 

Gcode shows the issue on every new layer. 

;MESH:NONMESH
G0 F300 X67.302 Y64.793 Z15.68 E1746.06377
G0 F15000 X67.262 Y64.753 E1746.06377
;TIME_ELAPSED:1485.634899
;LAYER:97
G1 F1800 E1736.73918 ; **Retraction (Correct and flow rate compensated)**
;TYPE:WALL-INNER
;MESH:accelerationtower.stl
G1 F1800 E1740.76579 ; **Prime ( Incorrect, not flow rate compensated )**
;FLOW_RATE_COMPENSATED_OFFSET = 5.769149
G1 F3863.3 X129.463 Y64.753 E1748.18999
G1 X129.463 Y65.149 E1748.20053
G1 X131.029 Y65.149 E1748.24219

**Expected results**

priming should add flow rate compensation so that the filament is flowing at the start. 

;MESH:NONMESH
G0 F300 X67.302 Y64.793 Z15.68 E1746.06377
G0 F15000 X67.262 Y64.753 E1746.06377
;TIME_ELAPSED:1485.634899
;LAYER:97
G1 F1800 E1736.73918 ; 
;TYPE:WALL-INNER
;MESH:accelerationtower.stl
G1 F1800 E1740.76579 ; **should be 4mm + a flow compensated amount**
;FLOW_RATE_COMPENSATED_OFFSET = 5.769149
G1 F3863.3 X129.463 Y64.753 E1748.18999
G1 X129.463 Y65.149 E1748.20053
G1 X131.029 Y65.149 E1748.24219

---------------------------
**Transition from inner to outer wall**
G1 X67.262 Y82.309 E1754.57483
G1 X67.262 Y64.753 E1755.04196
G0 F15000 X66.862 Y64.353 E1755.04196
;TYPE:WALL-OUTER
;FLOW_RATE_COMPENSATED_OFFSET = 5.365308
G1 F3863.3 X129.863 Y64.353 E1756.19711
G1 X129.863 Y64.749 E1756.20691

**Project file**
[accelerationtower.zip](https://github.com/Ultimaker/Cura/files/5081280/accelerationtower.zip)

**Log file**
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5081292/DxDiag.txt)

**Additional information**

if you enable flow rate compensation you get flow rate based retraction. I ran into an issue with this however as the retraction priming seems broken with flow rate compensation. The flow rate compensation is retracted after a move, however it is not re-primed which is an issue for the next move (especially in this case where the unbalanced extrusion move is quite long. ) the ending extrusion point is correct but the starting point is retracted

ClockeNessMnstr, 2020-08-17:
My Thoughts on this from #7329 

It's not working correctly but it was giving me increased extruder position.
you can check the gcode pretty easily. there are comments with "FLOW RATE COMPENSATION ="

however it is not using negative values or coming back to 0 when stopping before a retraction so the retraction occurs with the entire compensation value + retraction distance. since it doesn't reduce the compensation at the end of a move or before a turn it just extrudes more before the retraction. Then priming doesn't work because it does not restore the compensated position.

The description sounds like it should be breaking each move into 3 parts and adjusting the E values but I only see it adding some E at the first move after a feedrate change. It should reduce the compensation to 0 before the stop + retraction resulting in reducing the pressure to a consistant state before each retraction.

I'm going to break out a post processing script and see if I can replicate + improve on the changes it makes as I'm lacking Linear Advance on my Ender 3 V2 due to the integrated 2208 driver on my extruder.

What I think is going on:
Black: current behavior of Flow rate compensation / extrusion path of a line
Blue; target behavior
Red: normal behavior
Green: retraction

[image]

EDIT: E does not also represent pressure. maybe integral of pressure would look the same as E position.

Ghostkeeper, 2020-08-18:
The flow rate compensation adds or subtracts a bit of E to an extruded move based on the speed, line width and line length of the move. That's everything it does. Acceleration and deceleration of the trapezoid of a move are not taken into account, only the relative flow rates between lines.

This is more like what's happening (black is normal, magenta is with compensation):
[image]

The faster the extrusion must be, the more of an offset it'll have in the E value in the g-code. As a result, it's going to provide (theoretical) overextrusion if the flow rate becomes faster (a fast line after a slow line). It's going to provide underextrusion if the flow rate becomes slower (a slow line after a fast line).

What should happen then at the start of this bead, for the case of the start of the print and after every retracted travel move? The way that the implementation works now, and the way that I drew it, it starts with an offset of 0. Effectively the code is assuming that before printing the bead, it's starting off with a flow rate of 0. Fair assumption I'd say. The act of retracting completely disrupts the flow rate, so it'll need to build up pressure again anyway. The next move will always be overextruding since it's speeding up from 0 to the flow rate of that move.

If I read the code correctly, it seems that there is an exception for unretracted travel moves to keep the same E values throughout; it won't move the E during an unretracted travel move.

The retraction will be longer than usual when using compensation, since it's giving an overpressure in the nozzle throughout the entire bead. It's not actually adding any extrusion offset for the retraction or the unretraction.

ClockeNessMnstr, 2020-08-24:
I wrote a post processing script to do some LA

**Top: Slow, Middle: Fast, Bottom: Fast + SWLA**

[image]
[image]

Might not be perfect or efficient yet but it works really well since I'm stuck with TMC2208's on the Ender3 v2
[SWLA.zip](https://github.com/Ultimaker/Cura/files/5119260/SWLA.zip)

ClockeNessMnstr, 2020-08-24:
Ended up with a K_value of .355 on the bowden tube. Set Cura to 360mm/s which sets the print speed target at 180. 
Didn't check the actual speed but it was somewhere close to 2x

closed 2020-08-24 by ClockeNessMnstr