Ultimaker/Cura #8204: [4.7.0-beta] No extruder temperature in 'Monitor'

FelixJirka, 2020-08-17:
This issue seems to be related to #8176

**Application version**
4.7.0-beta

**Platform**
Win10 x64

**Printer**
V-Baby + V (running Marlin 2.0.6 on a SKR 1.4 Turbo)

**Reproduction steps**
1. Update from Cura 4.2.1 to current 4.7.0-beta
2. Attach Printer to USB 
3. Start Cura and open the Monitor Tab

**Screenshot(s)**
(1) PrintMonitor.qml & ExtruderBox.qml as shipped with 4.7.0-beta
[image]
(2) PrintMonitor.qml & ExtruderBox.qml taken from 4.2 branch
[image]
(3) Only PrintMonitor.qml taken from 4.2 branch & ExtruderBox.qml from 4.7
[image]

**Actual results**
Extruder temperature is missing in monitor, bed temperature is still shown

**Expected results**
Both temperatures are shown as in screenshot (3)

**Project file**
n/a

**Log file**
[2020_08_17_cura.zip](https://github.com/Ultimaker/Cura/files/5086129/2020_08_17_cura.zip)

**Additional information**
As you can see from the screenshots, I had some success with using files files from the old 4.2 branch. Maybe there is a new interface still missing in other code which the qml files are using?

I've also attached the log file for reference, but I think it is mainly complaining about my copy & paste actions of the qml files from the 4.2 branch. (and my other printers...)
Last start in the log file is vanilla 4.7.0-beta and second to last is with only PrintMonitor.qml taken from 4.2 branch and ExtruderBox.qml from 4.7.0-beta.

Thanks for your support!

fieldOfView, 2020-08-17:
This likely broke it: https://github.com/Ultimaker/Cura/commit/458acb356f11f50c3327b5af0be8531fb761548f

Ghostkeeper, 2020-08-18:
Yes, likely. I remember going through these ~50 calls to `extruders` and testing them all one by one, but maybe I skipped that one and just tested if the QML loaded properly since I don't have a printer here with USB printing that can actually display these extruders.

This refers to `PrinterOutputModel.extruders` rather than `GlobalStack.extruders`, so it shouldn't have been changed. I'll change it back.

commit dea53b9c (2020-08-18): Revert "Use extruderList rather than deprecated extruders property"

This reverts commit 458acb356f11f50c3327b5af0be8531fb761548f.
The change was wrong. The code refers to PrinterOutputModel.extruders which is not deprecated, not to GlobalStack.extruders (which is deprecated).

Fixes #8204.
closed 2020-08-18 by Ghostkeeper