Ultimaker/Cura #8205: [4.6.2] Support interface printed in midair using tree supports

darkpoetry, 2020-08-17:
**Application version**
4.6.2

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Sliced this figurine per the attached zipped .3mf file
[CE3PRO_Dwarf_Rogue.zip](https://github.com/Ultimaker/Cura/files/5086697/CE3PRO_Dwarf_Rogue.zip)
2. Noticed significant support interface being printed where there are no tree branches beneath them.

**Screenshot(s)**
[image]
[image]

**Actual results**
Support that becomes overextrusion zits on the the sides of the object

**Expected results**
No support interfaces should print without a support immediately below them.

**Project file**
[CE3PRO_Dwarf_Rogue.zip](https://github.com/Ultimaker/Cura/files/5086697/CE3PRO_Dwarf_Rogue.zip)

**Log file**
[cura.log.zip](https://github.com/Ultimaker/Cura/files/5086710/cura.log.zip)

Asterchades, 2020-08-18:
You should give the 4.7 beta a try - one of its main focuses is on tree supports. However, unless you have a specific need for the Support Interface (such as using a 2nd extruder with PVA so you can dissolve it) than I'd disable it entirely as it has issues with the newer tree supports, and possibly the older ones as well.

Unfortunately they're still not perfect in 4.7 as yet. But they're at least consistent now, so hopefully the next beta will show more improvements still.

jellespijker, 2020-08-18:
Thank you for reporting the issue. As @Asterchades mentioned the 4.7 beta has a lot of fixes for tree support. This is one of them. You could try out the new beta https://github.com/Ultimaker/Cura/releases/tag/4.7-beta

[image]

closed 2020-08-18 by jellespijker