Ultimaker/Cura #8206: Ender 3 V2 and filament change

Fabdu57, 2020-08-18:
Hi !

I have Cura 4.6.2
Firstly, ender 3 v2 is not present on printer lis.
Secondly, filament change script not works on V2. I ask to change filament at layer 30, i have totaly 300 layers; and printer continue to print without proposing me to change it...

Why ?

jellespijker, 2020-08-18:
It is correct that we currently don't have printer definition for the Ender 3 v2. We only have the time and resources to add Ultimaker printers, we rely on the community contributions for third-party printer definitions. According to this Reddit topic (https://www.reddit.com/r/ender3/comments/hg6ar0/ender_3_v2_cura_profile/ ) the profile for the Ender 3 can be used just as well.

In order to change the filament a `M600` ( https://marlinfw.org/docs/gcode/M600.html ) command is send to the printer. When trying the filament change postscript on on model sliced for a Ender 3, I see that it inserts a `M600` command at the correct position 
```gcode
M600 E30.00 L300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
```

I would suspect that the Ender 3 doesn't have native support for the `M600` commands.  But if you're feeling a little bit adventures you could try flashing Marlin on to your Ender 3, giving you a lot more options and it handles the `M600` commands correctly.
https://all3dp.com/2/ender-3-with-marlin-how-to-install-marlin-firmware-on-your-ender-3/

I will close this issue now, since the postscript works as expected.

closed 2020-08-18 by jellespijker

Fabdu57, 2020-08-18:
Ok thanks.
I found in marlin the filament change option. I try toactivate it and i will see...

dlopezpaz11, 2020-10-29:
@Fabdu57 could you tell me what you did to make the filament change??? please help me

scholi, 2021-03-19:
In marlin 2.0.7 you should edit Configuration.h and enable the `#define NOZZLE_PARK_FEATURE` feature (i.i. juste remove the two backslash chars at the beginning of the line).
Then in Configuratoin_adv.h enable `#define ADVANCED_PAUSE_FEATURE`