Ultimaker/Cura #8207: Cura prevents installation of other software.

achillesmjb, 2020-08-18:
**Application version**
(The version of the application this issue occurs with.) Cura 4.62

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Mac High Sierra OS v. 10.13.6 on a early 2011 Macbook Pro Intel Core I7.
**Printer**
(Which printer was selected in Cura?)
UM 2+

**Reproduction steps**
1. (Something you did.) Must delete Cura to install new applications
2. (Something you did next.)Reinstall Cura 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-08-19:
> Must delete Cura to install new applications

Can you please elaborate on this? Why must you delete Cura before installing new applications? What happens if you don't do this? What happens if you try to install other software while Cura is still installed?

achillesmjb, 2020-08-19:
Hello. Cura prevents new installations when I click on a newly downloaded program to install. When I click on the new file to open and install, Cura begins instead and opens just as if I clicked on Cura to open and use. The only way around this problem for me is to uninstall Cura then click on the newly downloaded program to install. After installation then I would go to UM website and reinstall Cura.

fieldOfView, 2020-08-19:
When you say "downloaded program", is that a disk image (*.dmg) or a package (*.pkg)?

achillesmjb, 2020-08-19:
Downloads were .pkg.

fieldOfView, 2020-08-19:
You could try [this article](https://www.laptopmag.com/articles/how-to-change-default-applications-mac) to set the default application for .pkg files back to "Installer".

Ghostkeeper, 2020-08-20:
I wonder how it managed to be registered to the .pkg extension though.

Cura is only able to open .3mf, .amd, .ctm, .dae, .glb, .gltf, .obj, .ply, .stl, .x3d, .zae, .gcode, .g, .png, .jpg, .bmp, .jpeg and .ufp files natively. No .pkg. As far as I know we don't specify that anywhere though except in the Windows installer.

closed 2020-08-26 by nallath