Ultimaker/Cura #8208: Save post processing scripts to print profiles rather than machine?

Motovader72, 2020-08-18:
It would be nice to save the post processing scripts within the custom print profiles rather than with the machine.

Scenario:
I've sliced multi color prints (change filament script) in a particular machine environment, 
then sliced a new solid color model with a different print profile in the same machine 
environment only to find I've forgotten about the scripts and the printer is stopping at 
the layers I used for the previous multicolor model.
 
I then have to delete all the scripts from that machine environment and reslice for solid color prints.
All my filament change layer numbers are then gone. I have saved custom print profiles for
various multicolor models that I have to r

***I understand I could create more machine environments to save individual scripts but
doesn't it make more sense to save the scripts to the print profile they're associated with?
Or at least have the option to choose? 

Thanks in advance for your consideration.

Ghostkeeper, 2020-08-20:
Architecturally, the built-in profiles can't be modified. They are stored in a location where Cura doesn't have write access (e.g. Program Files on Windows). So unless we make an architectural change there, you would only be able to save post-processing scripts if you use a custom profile, not with a built-in profile.

I can also see it go the other way around. You've added a post-processing script to replace all `M104` codes by `M109` because your printer firmware doesn't handle `M104` codes correctly. Then you're switching profiles, and forget that the post-processing script then gets removed too. And now your printer doesn't heat up correctly any more.

The way I see it, is that there are two types of applications for post-processing scripts:
* Some scripts apply to a specific print, e.g. Change At Z, Pause At Height, Filament Change. All of these are currently recognisable by that they are applied to a specific layer.
* Some prints apply to a printer, e.g. Retract Continue, Post Stretch, Use Previous Probe Measurements, Display Progress on LCD.

Some scripts can do both, like Insert At Layer Change, or Search and Replace. Currently we only really cater to the use case for the scripts that apply to a printer, by not automatically removing post-processing scripts when clearing the build plate or something. However there is a counter next to the slice button that indicates how many post-processing scripts are active, which should make it a bit more clear that there are still scripts active.

Motovader72, 2020-08-20:
I understand, thanks sincerely for your time and detailed response.

closed 2020-08-20 by Motovader72