Ultimaker/Cura #8209: Apply the bridge only to the layer directly above the support

viper0078, 2020-08-18:
I want to print only the layer directly above the support with a bridge.
Bridges can be made easier to peel off the supports by lowering the flow rate and increasing the printing speed.

A Bridge Skin Support Threshold of 100% will cause the bridge to be applied to layers other than the one directly above the support. 

[image]

The left image is as desired, but the center and right images are not the desired behavior.

Simplify3D will only apply the bridge to the layer just above the support.
I would like to see an option added to do the same thing.

Best regards.

viper0078, 2020-08-19:
The image is a slice of Simplify3D.

[image]

The bottom and top surfaces are not bridged.
Bridges are only applied to the areas that need support.

smartavionics, 2020-08-19:
Like I said on the forum, this feature (bridge over support) is available in my Cura builds.

viper0078, 2020-08-20:
Sorry. I didn't see all the options. Is this feature going to be built into UM BUILD?

viper0078, 2020-08-20:
As a side note, it would be nice if you could change the color of the bridge like in Simplify3D.

smartavionics, 2020-08-20:
> Sorry. I didn't see all the options. Is this feature going to be built into UM BUILD?

Probably not.

Ghostkeeper, 2020-08-20:
Well that's a bit of a lack of faith. If you don't intend to contribute your implementation, that doesn't mean that we can't build it ourselves.

The bridging functionality of Cura is on the weak side, with the experimental bridge settings only half done and the basic implementation hardly configurable at all, so some tweaking or tweakability on this front can't hurt. I'll see what my colleagues think of this feature request. I can see the advantage in bridging over supported parts, but we might need some different approach in how the direction of the bridge is determined.

mzhu1113, 2020-12-18:
I would like to see this implemented if at all possible as it'll reduce time spent having to tweak bridge settings and support coverage to trigger a bridge over support in our workflows as well as not having to use 99% Bridge Skin Support Threshold (and manual gcode editing) as a workaround to get the effects we want.