Ultimaker/Cura #8210: Assistance exporting node to STL file

noahgarfinkle, 2020-08-18:
I'm sorry if this has already been answered, but could not find in the existing issues or stack overflow. We are attempting to programmatically export our scene to an STL file, similar to how UM.OutputDeviceManager.requestWriteToDevice is called in the QML. We have gotten a bit lost in the code bases; is there a workflow to call the Uranium STLWriter from Cura? Our desired end state is to programmatically export the nodes in a scene to an STL file using a Python script within a plugin. Any tips would be sincerely appreciated, thank you!

rburema, 2020-08-19:
Writer-plugins register themselves with the kinds of files they can write.

I think you need to (indirectly) call `requestWrite` in `LocalFileOutputDevice` with the right mime-type, via a call to `OutputDeviceManager.requestWriteToDevice` with the `device_id` set to `"local_file"`.

nallath, 2020-08-19:
Alternatively, if you already have a stream you want to write to, you can use:
```python3
application = Application.getInstance()
mesh_writer = application.getMeshFileHandler().getWriter("STLWriter")
mesh_writer.write(stream, nodes)
```