Ultimaker/Cura #8211: [4.6.2] Surface mode appears to ignore Mesh Fixes setting

janherich, 2020-08-18:
**Cura version 4.6.2**

**Operating system Mac OS**

**Reproduction steps**
Surface mode appears to ignore Mesh Fixes settings, or at least `Maximum Resolution` and `Maximum Travel Resolution` and always generates move commands based on exact mesh/plane intersection.
I discovered this when my printer started to stutter/vibrate and print curved sections very slow compared to coarse grained polygons, but that happened only when model was sliced using the surface mode.
I can of course decrease the resolution of the generated model, but that is in many cases quite poor solution, because complex bezier/nurbs curves are usually approximated in non-uniform manner, eq more fine grained in more curved sections, etc.

**Screenshot(s)**
[image]
Comparing two G-code files of the exact same model (100mm long 10mm tall airfoil section with 200 points per each upper/lower surface), `wing_stub_normal.gcode` generated with normal mode, compared to `wing_stub.gcode` generated with surface mode, surface mode gcode is almost one order of magnitude bigger - ~3K vs ~27K lines) 

**Actual results**
Mesh fixes doesn't work with surface mode slicing

**Expected results**
Surface mode should respect mesh fixes settings, or at least display warning about ignoring them (the only `Affects` point mentioned is `Union Overlapping Volumes`)

rburema, 2020-08-19:
This happens because our current implementation of the simplify algorithm we use for (amongst other things) maximum resolution works on closed polygons, rather than open ones (like spirals). Also it should happen in another part of the program (after the  polygons have already converted to extrudable segments), which is another data-structure.

We discussed this, and unfortunately the effort required to fix this outweighs the priority of this issue, so, sadly, we chose to defer this for now.

smartavionics, 2020-08-19:
Surface mode does not ignore max-res/max-dev as you can see here...

Normal mode with fine and coarse res/dev...

[image]

[image]

Surface mode with fine and coarse res/dev...

[image]

[image]

smartavionics, 2020-08-19:
Maybe you are only altering max-res and not max-dev? Max-dev has far more of an influence than max-res does on the number of segments output.

janherich, 2020-08-19:
Hmm, this is strange, it just doesn't work for my model -> here is screenshot where I increase the max resolution/deviation to absurd high amounts and as you can see it only affected the skirt shape (which behaves correctly), but not the model:
[image]
This is the actual STL model I used for slicing -> https://gist.github.com/janherich/8add63127438cc6131df1ad786956c2d

smartavionics, 2020-08-19:
You're right, for that model, max-res/max-dev are ignored. I shall look into that...

smartavionics, 2020-08-19:
Solved it! the problem is with the model, its normals are wrong. If you correct the normals, you get the expected output...

[image]

janherich, 2020-08-19:
Awesome, thanks @smartavionics ! Yes, the airfoil has one surface pointing in the wrong directions (normals are reverse), I suspected that it wouldn't matter with surface mode, but apparently, it does (@rburema provided explanation why, mesh fixes work on closed polygons representing slices).
Unfortunately it doesn't fix all situations, because sometimes, the polyline is not closed.

smartavionics, 2020-08-19:
But the mesh tools plugin fixes the normals just fine.

smartavionics, 2020-08-19:
The polygons do get closed when the normals are correct.

janherich, 2020-08-19:
Correct, but imagine model like smoothly corrugated surface -> in x/y plane cut, it's just one open polyline.
Surface mode is great because you can easily create such mesh and not worry about generating correct thickness, etc, as it will be always nozzle width. 
Unfortunately max resolution/deviation doesn't work in that case.

smartavionics, 2020-08-19:
That's true, I'll ponder that.

TheRooster13, 2021-11-09:
Being able to adjust resolution on non-closed meshes would be a HUGE benefit for me. As it is, the prints take too long to be practical because of small features in the mesh.

closed 2023-06-13 by github-actions[bot]