Ultimaker/Cura #8212: [Cura 4.7 Beta] Crash Report

DivingDuck, 2020-08-19:
**Application version**
Cura 4.7 Beta

**Platform**
Windows 10, Radeon RX 580 series

**Printer**
Own build, AGeHEVO5

**Reproduction steps**
1. (Something you did.)
I activate a second extruder in Cura. After reloading Cura everything works fine so fare
2. (Something you did next.)
I open an older project file with only one extruder activated and had a first crash. After that I was asked to send a crash report what I did. I don't make a cleanup, instead I reload Cura again and it was working. But now I have the crash again with starting Cura

**Screenshot(s)**
n/a, but see attachment

**Actual results**
Cura crashes more often. Pls. see attachment.

**Expected results**
No crash of cause

**Project file**
n/a

**Log file**
[Cura4.7CrashReport.zip](https://github.com/Ultimaker/Cura/files/5096006/Cura4.7CrashReport.zip)

**Additional information**
This is my printer definition 
[AGeHevo5.zip](https://github.com/Ultimaker/Cura/files/5096062/AGeHevo5.zip)

rburema, 2020-08-20:
I have to prepare for a meeting now so I don't have the time to actually test it right this minute, but I did see that `"machine_extruder_count"` is missing from the definition, which should cause it to fall back to the default in `fdmprinter.json`, which is a count of 1, while the machine has 2 extruders defined.

While this _probably_ still shouldn't cause a _crash_ I don't expect it to work properly for this scenario (mismatch between actual defined extruders and 'extruder count').

fieldOfView, 2020-08-20:
The custom fff printer defines 8 extruders, but `machine_extruder_count` can be set to any number >=1, <= 8.

Ghostkeeper, 2020-08-24:
It's crashing because the setting version changed. Cura doesn't upgrade the types of profiles that can't be modified through its interface, in particular the nozzle profiles you modded in. You'll need to upgrade the nozzle profiles manually. Basically that involves updating the `setting_version` and making the necessary changes to tree support settings if you have those.

It'll load the nozzle profiles correctly then. It's crashing on a line where it asks the `VariantNode` (representing a nozzle profile) which materials are available for that combination of nozzle and printer. Since your printer says it has nozzle profiles, it assumes that there are nozzle profiles. But since there are no actual loaded profiles, this somehow ends up on None.

This is a mistake on the side of the mod, so we're not going to fix it. While we try to be somewhat robust and stable against external changes, we can't foresee any situation so it's out of scope to be able to deal with broken configuration folders like this.

closed 2020-08-24 by Ghostkeeper

DivingDuck, 2020-08-24:
Well, let me state, I am impressed and amazed. Your company ask for testing a new software release. Anyway, sorry for pointing to this problem and thanks for taking the time.

Ghostkeeper, 2020-08-25:
Yeah, sorry. We're aware that you can modify Cura to the point that it breaks. It's the flip side of having moddable software like this. There are much worse ways in which you can break it too, e.g. by adding shaders that crash the GPU and such. We consider it out of scope. If you modify the files in Cura's resource folder manually, you're on your own.

Profiles that are submitted (via pull request) to be included with Cura's releases will be maintained by the developers themselves though. That can be a consideration too.