Ultimaker/Cura #8213: 4.6.2 - Prime Tower using disabled extruder / non prime tower

nkatsoulis, 2020-08-19:
**Application version**
(The version of the application this issue occurs with.)
4.6.2
Windows 2010
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Diabase H Series 5 tool heads
(Which printer was selected in Cura?)
Custom FFF
**Reproduction steps**.

I have extruder 4, 5 selected on all models and the Prime Tower wants to use a different extruder. I am not finding where to set this in advanced custom settings or right-clicking the tower? When I enable extruder 1 I can slice but I do not have anything that I can see for the prime tower selected for extruder 1, Also when I view materials it is showing as extruder 2 in the prime tower area. Thoughts?

(Image showing the problem, perhaps before/after images.) 
[image]
[image]
[image]

Photo without prime tower and Extruder 1 has to be enabled 

[image]

**Actual results**
(What happens after the above steps have been followed.)
Extruder 2 is being used for the prime tower when it is disabled and will not let you slice if it is disabled 

**Expected results**
(What should happen after the above steps have been followed.)

Prime tower should not use disabled extruders to prime

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

No Prime Tower with extruder 1 enabled

[NoPrime Tower.zip](https://github.com/Ultimaker/Cura/files/5099645/NoPrime.Tower.zip)

Prime Tower with Extruder 1 Enabled

[withPrime Tower.zip](https://github.com/Ultimaker/Cura/files/5099648/withPrime.Tower.zip)
[image]

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-08-20:
Still happens in master as well. I'll bring it up with my colleagues.

martijnschouten, 2020-09-14:
I ran in the very same issue today:

**Application version**
Cura 4.7.0

**Platform**
Windows 10

**Printer**
Diabase H-series

**Reproduction steps**
- Slice a 2 material print on a 5 material printer with the first 3 materials disabled and PLA selected in the first (disabled extruder)
- Create a prime pillar

**Screenshot(s)**
[image]

**Actual results**
Printer uses one of the disabled extruders in the prime pillar instead of one of the enabled extruders.

**Expected results**
The printer uses only the two enabled extruders. 

**Project file**
[3mf files.zip](https://github.com/Ultimaker/Cura/files/5216596/3mf.files.zip)

**Log file**
[cura (2).zip](https://github.com/Ultimaker/Cura/files/5216599/cura.2.zip)

**Additional information**
The error does not occur in case I have DSM novamid 1070 selected instead of PLA for the first extruders.

Ghostkeeper, 2022-02-15:
Shouldn't that just get deferred then? It is still an issue for people.

fvrmr, 2022-02-25:
Changing this to deferred instead of won't fix. Anyone is free to fix this issue and create a pull request.

closed 2023-06-13 by github-actions[bot]

lrcg, 2024-02-05:
Workaround: disable the unused extruders before slicing