Ultimaker/Cura #8224: Expose disallowed print areas to UI

derekantrican, 2020-08-23:
**Is your feature request related to a problem? Please describe.**

There are many reasons to want disallowed print areas on a printer bed. Maybe there's clamps/clips there, maybe there's a gouge in the bed (my case), or maybe there's a previous print still on the bed that should be avoided.

**Describe the solution you'd like**

In the best case, I'd like to be able to add "disallowed print areas" the same way that I can add support blockers - just click and place. But in a minimal case, it'd be helpful to expose the disallowed print areas to the UI (probably in the printer-specific settings) so that they can be customized. In addition - it seems that Cura adds a ~10mm margin to disallowed print areas and this can be an issue if you're just trying to avoid a very small area (eg 5mm x 5mm)

**Describe alternatives you've considered**

I've tried editing the printer .inst.cfg file manually and that works...kinda. It requires Cura to be closed, then to relaunch Cura. Then it adds a ~10mm margin around the disallowed area that I don't want, and it's also difficult to measure the bed manually then adjust the coordinates for Cura (Cura works based off the origin of the bed, not from the side and it seems that Cura considers the back negative and front positive - against what I would expect for the coordinate system). All-in-all, it's just not easy or intuitive.

**Affected users and/or printers**

I think a lot of people could benefit from this - particularly anyone that has the situations I mentioned at the top

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-08-24:
> In addition - it seems that Cura adds a ~10mm margin to disallowed print areas and this can be an issue if you're just trying to avoid a very small area (eg 5mm x 5mm)

It does this because your model may be enveloped with a brim, or a raft, or it may need to move around your model to avoid hitting it with a travel move. Or you have Horizontal Expansion. Or a draft shield, or an ooze shield. There are a number of settings that cause it to draw disallowed areas bigger.

We could expose the setting that determines this to the interface. But that would remain a text box then where you need to manually enter coordinates, separated by commas and brackets correctly. To make it any more intuitive would require some graphical interface where you can draw polygons on the build plate or something, and that is quite a complex feature to ask for, both to develop and for a user to use.

The setting is a printer setting, which is why it gets hidden. We consider the disallowed areas to be inherent to the printer hardware, so changing them equates to changing the printer hardware, which is developer-level expertise so you'd be able to adjust this JSON file accordingly then. The use case of wanting to avoid a scratch or a previously printed object wasn't considered here though. The Printer Settings plug-in on the Marketplace also doesn't expose the setting even. Probably because it doesn't have an interface element for polygons that fits in the settings list. It would need to add something.

derekantrican, 2020-08-24:
Yeah, I understand why there's a margin and totally understand that this should be equated to a printer hardware change. I think exposing some textboxes (both for the coordinates and for the 10mm margin - maybe even a margin specific to each disallowed area?) in the printer settings ui would be sufficient for at least modifying it easier (especially if someone is adding some sort of special/custom clips that would require modifying this). Maybe this could be added as a first draft then a graphical interface could be backlogged as a future ask?

An additional, related thing (I can make this a separate issue) - it seems that disallowed areas get saved with a project. Which means that if I save a project, add a disallowed area, then open the project it will clear the disallowed area.

Ghostkeeper, 2020-08-26:
> An additional, related thing (I can make this a separate issue) - it seems that disallowed areas get saved with a project. Which means that if I save a project, add a disallowed area, then open the project it will clear the disallowed area.

The disallowed areas are a setting like any other and thus get saved with the project files, yes. It depends a little bit on how you modified it though because the .def.json files don't restore from the project files, so modifications there should not get changed.

There is a bit of an aversion against leaving features in a half-finished state. I know that Cura has rather many of those too (e.g. the experimental section of settings) but it has something unprofessional and cluttering about it. While it may be better for those that would use the unfinished feature, it makes the user experience worse off for those that don't use it. So we'll defer this until there is more room in the priority list or until there is more of a reason for users to customise this.

closed 2023-07-13 by github-actions[bot]