Ultimaker/Cura #8229: [4.7.0] Profiles missing and cannot be created

kelaymon, 2020-08-25:
**Application version**
4.7.0

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
Opened Cura and checked for profiles. My default 0.2 mm profile settings are there but when I click on "Update profile with current settings", it appears to not create a profile.

**Screenshot(s)**
[image]

**Actual results**
"Manage profiles" shows no entries either for previously created profiles in 4.6.2 or for the newly updated settings.

**Expected results**
Profiles should be there

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/5123749/cura.log)

**Additional information**
Upgraded from 4.6.2

kelaymon, 2020-08-25:
After restarting the program several times, now my profiles are showing up. Nothing additional was done.

nallath, 2020-08-25:
Do you perhaps have a custom material active? Cura 4.6.0 used to have a bug where the wrong user profiles would be visible. Since that has been fixed in 4.7.0, that might be what you're seeing.

kelaymon, 2020-08-25:
Yes, I have 4 materials created due to slight differences in print temps and diameters. One of those was active when all this happened, but it's still active and has not been changed. I restarted Cura about 6 times during all this and it finally started displaying profiles. In the image below, the items dotted red were all missing. The blue dotted item was there but would not appear in the profiles list in the settings.

[image]

nallath, 2020-08-25:
I don't see a whole lot in the logs. Could you share your 4.6.2 configuration folder?

jjtk88, 2020-08-25:
FWIW, I have this problem as well.  I just upgraded from 4.6.1 -> 4.7.0.   I have a number of custom profile and materials created.  When selected, all of my profiles disappear and am unable to select any old profiles.  The Profile selection drop down just says "Not Supported"

[image]

[image]

kelaymon, 2020-08-25:
What is the typical path to the folder you need? I have a 4.6 folder under %appdata%\cura but when I try to zip it up to send it, it's 26 MB.

jjtk88, 2020-08-25:
Wanted to add one more comment.

The material I had loaded when I initially launched Cura 4.7 seems to still be there, and Cura auto-selects "Standard Quality" by default.  However when I select any other Material from the dropdown, the profile menu options disappear.

[image]

kelaymon, 2020-08-25:
So I have a similar issue to jjtk88. I have 4 custom materials. When I select 2 of them, it appears that my profiles are working including the defaults. When I select 2 other ones, I get the "Not supported" message as well and no profiles are shown, not even the built in defaults.

nallath, 2020-08-25:
That's probably because those materials have a different type. So if you changed your custom PLA to be of type PLA+, it doesn't have built in default ones. It was actually a bug in previous cura versions that it was showing them.

If you can send me the config folder (either by stripping out the plugins or just uploading it somewhere), I can verify if this is the case or not.

jjtk88, 2020-08-25:
@nallath ah, I see that now.  I changed my "PLA2" override -> "PLA" and it seems to all work now.

Is it not possible to specify material types other than the "common" ones?

kelaymon, 2020-08-25:
https://drive.google.com/file/d/1naxC-B_bht6KmETrS47-gWeHoLwcMFK_/view?usp=sharing

I modified the Material Type field from "PLA+" to "PLA" and that seems to have resolved it.

nallath, 2020-08-25:
How it should work:

1. Select Generic PLA
2. Open profile dropdown window; notice a number of profiles (Super, dynamic, standard, low, & custom profiles).
3. Create a new material
4. Change type of the material to "FOO"
5. Material selection becomes yellow (since there isn't a quality profile for a material of type "FOO"
6. Only profiles created on this not-supported state should be visible.

It is possible to create new types, but we also need to know what qualities are actually supported by that material. For a slew of materials we have some pretty sane defaults. But it uses the type to figure out with what it needs to match.

So it's perfectly okay to make "My custom PLA", as long as its _type_ is PLA (so that Cura knows what material category it is!). If it doesn't find it, well, it just goes in this "warning" state (telling you: Hey, you can play around with settings, but the defaults for this _might_ not be good).

jjtk88, 2020-08-25:
I never would've thought *that* was the reason why the material showed up as yellow!  I always noticed that a handful of my profiles showed up as yellow, but after poking through the "Print Setting" menu I could never find any setting that appeared to cause the material to be yellow.  I (erroneously) figured that the "material" input box was there for tagging/organization purposes.  This solves a mystery for me!

As an aside, would a drop down menu be more appropriate under "Material Type" to make it more obvious?

kelaymon, 2020-08-25:
No problem. Not having the "+" on the end of it is not the end of the world. I can just put it in the Display Name field. The only reason I put it in there to begin with is Inland has PLA and PLA+.

I agree with jjtk88 that a drop down for material type would be nice if it's going to be a field that is expecting a certain input.

nallath, 2020-08-25:
> I (erroneously) figured that the "Material Type" input box was there for tagging/organization purposes. This solves a mystery for me!

That's what the label is for. But yeah, we probably should think about how to fix this.

> I agree with jjtk88 that a drop down for material type would be nice if it's going to be a field that is expecting a certain input.

We've been debating solutions for this for quite some time now. I don't think anyone came up with the dropdown. I sure as hell feel a bit stupid for not coming up with it.

RaabenF, 2020-08-28:
> How it should work:
>
> 1. Select Generic PLA
> 2. Open profile dropdown window; notice a number of profiles (Super, dynamic, standard, low, & custom profiles).
> 3. Create a new material
> 4. Change type of [...]

I still dont get it :(

is this the only workaround? im going back to 4.6.2

nallath, 2020-08-28:
It's not a workaround. The situation in 4.6.2 is flat out wrong. You will not get the correct settings (or settings that depend on what configuration you had previously active).

RaabenF, 2020-08-28:
thanks @nallath 
i see, but it worked. so there's actually no way migrating the old profiles? i had custom materials created, that shoud mach my print profiles, but still only the PLA based ones are visible.

sry didnt even knew, that cura is opensource, is it? can one contribute, i dont see stuff in the wiki about that

Ghostkeeper, 2020-09-01:
> BTW didnt even know that cura is opensource, is it? can one contribute, i dont see stuff in the wiki about that
thanks for the response anyway. cluddering configurationthings hmm... thats how it is

Cura is open source (because you can [see its source code](https://github.com/Ultimaker/Cura/blob/master/cura/CuraApplication.py)) and free (as in "freedom", because you can [contribute](https://github.com/Ultimaker/Cura/pulls)). Although free software is not black-and-white, we do our best to incorporate contributions through bug reports, feature requests and pull requests, and Cura's license allows you to make modifications and publish them, which makes Cura "free software". That doesn't mean that it's easy to make code changes though. Cura is still a very big system, with a complex structure, especially the profile system.

> i see, but it worked. so there's actually no way migrating the old profiles? i had custom materials created, that shoud mach my print profiles, but still only the PLA based ones are visible.

The easy workaround is to change the material type to something that has supported quality profiles.

Cura can't migrate these profiles automatically because it's missing information. In particular, these profiles don't have the information as to which built-in quality profile they were based upon. In Cura 4.6, we were not filtering custom profiles correctly, which made custom profiles appear regardless of what built-in profile they were based on. So even though a profile was based on the "draft" profile, it would appear in the list of profiles even if you're using a small nozzle size and there is no "draft" profile for that nozzle size. It would select an arbitrary profile then to base it on, which causes the print to be completely different.
Now in 4.7, this filtering is fixed. The profile no longer appears in the list if it's based on a quality level that's not available. But if you were using the profile in that incomplete state it causes a workflow issue. The profile still stores that it's based on the "draft" profile, so it won't appear in the list if there is no built-in "draft" profile.

You could repair the old profiles manually in your configuration folder, which would cause the profile to finally become stable, actively marking it as being based on an empty quality profile. This involves making changes to the configuration folder though, so it's easy to make mistakes and you should make a back-up first.
What you would need to do is to go to Help -> Show configuration folder and close down Cura. Then search for the `quality_changes` subfolder and in there find all files that have a name similar to the profile you want to edit. There should be one file for each extruder that your printer has, and one extra file for the global settings. In each of these files, under the `[metadata]` tag, search for the `quality_type` line and change its value into `not_supported`. Save the files, then start up Cura. They will no longer be available for printer configurations that have supported quality levels, but they will then be available for those that don't.

danielrmatos, 2020-09-04:
I'm having the same problem changing nozzle size. If I put 0.6 or 0.8, profiles are missing.

nallath, 2020-09-07:
> I'm having the same problem changing nozzle size. If I put 0.6 or 0.8, profiles are missing.

That's because not all profiles are suited for 0.6 or 0.8 nozzle.

danielrmatos, 2020-09-07:
I don't think this is the problem. Even default profiles are missing.

nallath, 2020-09-07:
> I don't think this is the problem. Even default profiles are missing. 

Do you have a custom material selected? Because that is usually what causes this (and also what is explained in this thread to be the problem)

danielrmatos, 2020-09-07:
No, any material.
I'm looking in the profiles files and found nothing related to nozzle size.
But if I select 0.6 or higher, appears only a "Draft" profile with 0.32
layer height that is not visible in 0.5 and lower.
So I changed a profile to 0.32 layer height and... Nothing, it not showed
in 0.6 or higher.
I changed the draft profile in 0.6 (just changed the layer height, to 0.2)
and save as new, but nothing, this "new" profile does not appear.
And another thing is sometimes when I change nozzle, different profiles
appears or vanish.
Now, for example, I changed to 0.6 nozzle and instead of "Draf 0.32" showed
only "Standard Quality 0.2" (few minutes ago, only Draft 0.32).
This is very strange. :)
I'm using Cura 4.7 with Linux (Ubuntu 18.04).

nallath, 2020-09-07:
The behavior that you are seeing is entirely correct. If you create a new profile, it's based on an existing profile (eg; if you create a profile when 'draft' is active, it's based on draft).

If you then change the layer height to 0.06, it will still be based on "draft". If a certain material / nozzle combination doesn't have draft, your profile will not be visible (since it doesn't exist for that combination).

danielrmatos, 2020-09-07:
First of all, thanks for your answers.
When I select 0.6 nozzle change a profile and save as new, these new
profile are not shown (they just disappear, and sometimes works as
expected).
While I'm writing, I tried to reproduce and this time 3 profiles appears in
0.6 nozzle: standard 0.2, low 0.28 and draft 0.32.
It's completely random what profiles will appear or not. This is not a
expected behavior.
I'm just trying to understand what to expect to save my profiles and avoid
reworking.
I'd like to understand what make a profile appear on a nozzle size and not
in another one, I found nothing inspecting a profile file.

nallath, 2020-09-07:
You're going to have to give a bit more information if you want me to reproduce it.

So; what printer do you have and provide a step by step list of actions that you did
(eg;
1. Select config x 
2. Change setting
3. Create new profile via dropdown
4. Check in dropdown if profile is present, etc)

tvercruysse, 2020-09-14:
> The behavior that you are seeing is entirely correct. If you create a new profile, it's based on an existing profile (eg; if you create a profile when 'draft' is active, it's based on draft).
>
> If [...]

I understand that this is correct from a programmers point of view, however this behavior could be improved.
Whilst i think the change is a positive one, the old custom profiles should not have been affected by this change.
i understand wanting to hide the default (incompatible ??) profiles - although i am perfectly capable of printing up to as low as 0.15mm/layer on a .6 nozzle - hiding user's custom profiles is confusing at best, where there always were profiles when switching nozzle sizes, now there suddenly aren't. Especially since i had cura setup on a 1mm nozzle by default, it looked broken.

Keep up the good work.

Ghostkeeper, 2020-09-15:
We're clarifying the case of importing a profile that's not currently supported, for 4.8.

We tried to keep the old profiles visible, but Cura doesn't have enough information to change the profiles automatically to rebase them on profiles that are supported.

reynol3232, 2020-09-18:
I still don't understand.  I'm sorry.  I upgrade from 4.5 to 4.7 and lost many profiles.  If a profile is connected to a custom material how the heck can I see what material the profile is connected to so that I can select that material so that my profile is visible again?  (I think I'm explaining this correctly)
This is a screen shot of the profile in 4.5 that I want to show up ion 4.7.  I've tried exporting from 4.5 and into 4.7.  I give up.
 Please help
[image]

msteele999, 2020-09-20:
I had this issue as well and noticed that in Cura 4.6.1 the Nozzle size appeared to NOT be tied to the printer, it was displayed in the main display window. In Cura 4.7.x it is down in the printer configuration / extruder configuration. If I created a new printer in 4.7.x, all my materials no long reported as 'unsupported'. This only occurred on imported printers from 4.6.1

There were built in system materials in 4.7.x that would report this way as well.

[image]
[image]
[image]

Ghostkeeper, 2020-09-23:
> I still don't understand. I'm sorry. I upgrade from 4.5 to 4.7 and lost many profiles. If a profile is connected to a custom material how the heck can I see what material the profile is connected to [...]
I've tried exporting from 4.5 and into 4.7. I give up.
Please help

Your custom profile is not connected to a particular material. It's based on a particular quality level. You can only use it with a material that has profiles for that quality level. The quality level is the built-in profile, what most people see as "layer height". Except you could also change the layer height in a profile of course, and there are more settings that usually change with the quality level.

So to make it work you will need to change the material type of any custom material to a type that Cura does have profiles for. That's the field selected here:
[image]
You can still create custom profiles for material types without built-in profiles, but only by creating new profiles in 4.7 or later.

murk-sy, 2020-09-25:
**Workaround that worked for me:** 
In the folder ``%appdata%\cura\4.7\quality_changes``: in each file, find the line containing ``quality_type = <not supported/draft/etc>`` and replace it with ``quality_type = standard``. This might break things but it probably won't.

I've found it especially annoying to start Cura, find some profiles missing, and then restart it only to see *even more* profiles missing. Honestly I can't even say I'd be able to reproduce that. 

Broken profiles should be at least listed inside the program or a notice should be given "some profiles weren't loaded because they are incompatible" when opening the profiles menu. Anything other than just a blank list would be an improvement.

fabriceimperial, 2020-09-26:
After looking at that "not supported" mess my first impression is there is no UI/UX engineer nor QA working in this project.

For example on my version (4.7.1) the following behaviour is so messed up and those are not custom materials and not even a popup or a sanitized list of internal materials:

- "generic CFE" type "CFE" -> NOT working
- "generic HIPS" type "HIPS" -> NOT working
- "generic Nylon" type "Nylon" -> NOT working
- "generic PC" type "PC" -> NOT working
- "generic PVA" type "PVA" -> NOT working

- "generic TPU 95A" type "TPU 95A" -> working  <---------
- "eMotionTech TPU-98A" type "TPU" -> NOT working  <---------

- "eMotionTech ASA-X" type "ASA-X" -> NOT working
- All the eMotionTech soluble (BVOH, HIPS, PVA-M, PVA-S) type "Soluble" -> NOT working
- "Leapfrog Natural PVA" type "EPLA" -> NOT working
- "REDD ASA" type "ASA" -> NOT working
- "REDD HIPS" type "HIPS" -> NOT working
- "REDD Nylon" type "Nylon" -> NOT working
- "REDD TPE" type "TPE" -> NOT working
- "TiZYX FLEX TiZYX" type "FLEX" -> NOT working
- "TiZYX PVA TiZYX" type "PVA" -> NOT working
- All the Velleman PLAs type "PLA-XXXXX" -> NOT working
- All the eSUN PLA+ type "PLA+" -> NOT working

nallath, 2020-09-28:
> After looking at that "not supported" mess my first impression is there is no UI/UX engineer nor QA working in this project.

Thanks for the constructive feedback. This is sure to improve things ;)

fabriceimperial, 2020-09-28:
> > After looking at that "not supported" mess my first impression is there is no UI/UX engineer nor QA working in this project.
> 
> Thanks for the constructive feedback. This is sure to improve things ;)

Of course it is. Because there is a company name behind it and customer/user frustration can have a significant impact on business and it is hard to make a second first impression.

And UX, QA and Dev are 3 different trades.

JBollow, 2020-09-29:
i just came across this problem myself, after an update my old profiles werent showing up.
There seems to be an issue with the [metadata] in our profiles.

**To solve this issue**, go to: AppData\Roaming\cura\4.7\quality_changes

Open your cfg pair (one printer one extruder) and change the [metadata] to:

for base_extruder:
[metadata]
intent_category = default
quality_type = standard
position = 0
setting_version = 15
type = quality_changes

for printer:
[metadata]
quality_type = standard
setting_version = 15
type = quality_changes

Your not working profiles probably have: quality_type = draft
I am not sure why this changed or is not working any more, but for me it fixed the problem

nallath, 2020-09-29:
> Of course it is. Because there is a company name behind it and customer/user frustration can have a significant impact on business and it is hard to make a second first impression.

The profiles that go wrong are contributed by the community and are of a competitor. You can't expect us to fully test and guarantee that these things work. It's already a boon that the software is provided as is, without us having to sink even more resources into this. We guarantee that it works with our products. 

Anyhow, giving feedback is fine, but I'm getting real tired of all the verbal abuse. Telling us a feature isn't working is fine, but just stop with calling people incompetent. It's not making the problem go away. If anything, it makes it less likely. 

> I am not sure why this changed or is not working any more, but for me it fixed the problem

It makes it _look_ like the problem is solved. What is happening is that your profile is combined with another profile. This means you are likely to get different settings.

fabriceimperial, 2020-09-29:
> The profiles that go wrong are contributed by the community and are of a competitor. You can't expect us to fully test and guarantee that these things work. It's already a boon that the software is [...]

Maybe I'm not clear but I understand your technical point of view, what I'm saying is: from a basic user point of view if a feature breaks without any hint it is not well perceived.

So here are my questions:
- If those external profiles are not passing filters why are they still enabled/displayed? (because a user can't even modify them in the UI to change there type, at least I cannot).
- Why a generic TPU material type is "TPU 95A"? Are those generic materials also from the community?

> Anyhow, giving feedback is fine, but I'm getting real tired of all the verbal abuse. Telling us a feature isn't working is fine, but just stop with calling people incompetent. It's not making the problem go away. If anything, it makes it less likely.

I have never called anyone incompetent, if it is how it sounds like I apologize, I'm talking about process, methodology and room for improvement. I know it is hard to keep a standard/global user vision when you work on a project. That is why a PM is welcome.
In the end, I don't know how your team is organized, it is just how I see it from the outside.

Ghostkeeper, 2020-09-29:
> If those external profiles are not passing filters why are they still enabled/displayed? (because a user can't even modify them in the UI to change there type, at least I cannot).

We leave it up to the printer profile to choose which materials are displayed and which are hidden. By default we display a material but give a warning colour to the user if there are no profiles for it. That way a user can still use the profile but knows that the settings are not optimised. The user can then still optimise their settings and create a profile for it himself. If the material is hidden, the user can't create a profile for it either, and in fact can't create their own material profiles with new material types at all.
Some printer definitions choose to hide materials that they know don't work well with their printers at all. For instance, [none of the Ultimaker printers support silicone or the HIPS/PETG materials](https://github.com/Ultimaker/Cura/blob/1ac59d88570ef6938d63377afa5d00e82928155e/resources/definitions/ultimaker.def.json#L9), and [HMS434 excludes pretty much all materials, claiming that it doesn't work with their printer](https://github.com/Ultimaker/Cura/blob/1ac59d88570ef6938d63377afa5d00e82928155e/resources/definitions/hms434.def.json#L12).

> Why a generic TPU material type is "TPU 95A"?

I consider that technically a mistake. Generic TPU should include all flavours of TPU that are close enough that they can generally work with the same profiles without sacrificing print quality too much. 90 or 95 probably doesn't matter too much, so in the label it should probably not specify 95A. And if it does matter, the profile ID is wrong since that doesn't specify 95A.

> Are those generic materials also from the community?

Many of them are. All of the 1.75mm variants are contributed by the community since Ultimaker doesn't have printers that support 1.75mm filament. Generic PETG and HIPS were also provided by the community in 2.85mm flavour. They serve as a good starting point for other profiles of specific brands.

> I have never called anyone incompetent, if it is how it sounds like I apologize, I'm talking about process, methodology and room for improvement. I know it is hard to keep a standard/global user vision when you work on a project. That is why a PM is welcome.

It felt a little passive aggressive. We of course do have a UX designer and multiple QA engineers.

It's just out of our power to ensure that all printers have profiles for all combinations. If we get a new material by Fabtotum which someone supplies because they have a profile for their Ender 3 for it, should we reject it because the uBuild-D Mr Bot 280 doesn't have a profile for it yet? It's untenable. We'd like to keep power to the user to create profiles so we must allow the user to choose it. We just warn them for unsupported combinations.

If you know of a way to make the user experience better, that would be constructive criticism.

reynol3232, 2020-09-29:
With all of the hassle I tried PrusaSlic3r and I really like it! I've used Cura since version 2.X. unfortunately Cura lost me.

fabriceimperial, 2020-09-30:
> We leave it up to the printer profile to choose which materials are displayed and which are hidden. By default we display a material but give a warning colour to the user if there are no profiles [...]

Ok, so that is a huge piece of information.
Looking at it, it seems only internal, because even if my definition folder exists (~/.local/share/cura/4.7/definitions/) it is empty could it be because I have switched to AppImage builds?

> It's just out of our power to ensure that all printers have profiles for all combinations. If we get a new material by Fabtotum which someone supplies because they have a profile for their Ender 3 [...]

So here is what I see as a basic user (why I was looking for a problem and the chain of events):

First launch (and first time I use a 3D printer that I haven't built myself) before the patch:
- Start Cura.
- Select an existing printer profile.
- Heavily modify it to fit my needs and save it (via Create profile from current settings/overrides…).
- Create TPU Material for fun because to me it is just an easy way to change hotend/bed temperatures and retraction setting and it avoids me to create too much custom profiles (by the way I think the dropdown width is way too small for the profile and way too large for the material considering the amount of important information ellipsised in the profile in the main UI).
- Everything is fine for a while (I only have PLA and TPU at the moment).
--> Then the patch comes
- And I switch back to TPU for a project in the main UI and the material field background is now colored.
- I don't understand why and the tooltip only display the name of the material, so no hint here.
- The Profile field displays a Not supported text and fallback on I don't know what because all my settings are now changed and all the profiles disappeared from the list, so to me: "WTF is going on, why my profile is broken?"
- So I go to the Preferences>Profiles and the Printer field display something but the list is blank: "Fuck I just lost all my profiles (that took hours to make) and I don't know if my exports are recent/good enough"
- I import my old saved profiles. Still nothing, the list is still empty: "Maybe I need to revert to previous Cura version? I hope nothing is lost…"
- Before doing that I go to the Preference>Materials.
- Here NO background colored material in the list to show that something is wrong but you can "Activate" a material.
- I activate one by one every materials and look at their info and print settings to try to find a pattern.
- OK PLA seems to work but a lot of others do not and even Generic ones have issues (to me Generic materials appear, at the moment, to be fully internal and the absolute truth).
- And I finally, look at the Generic TPU and see that it has a type TPU 95A so I put it in my custom profile and BAAAM it works!
- And even the previously imported Profiles are now visible.

In the meantime, between those lines I have a web browser open to search for similar problems. So I think that you could understand why it lead me to those quick conclusions.

Ghostkeeper, 2020-10-02:
> Looking at it, it seems only internal, because even if my definition folder exists (~/.local/share/cura/4.7/definitions/) it is empty could it be because I have switched to AppImage builds?

No that's normal. Cura has two resource locations, one in its actual application and one in your `.local` folder, with identical folder structures. There will be definitions inside the `definitions` folder inside of the AppImage. The alternative locations is in `.local` so that you could theoretically mod additional definitions there that you find on the internet or create yourself.

> (by the way I think the dropdown width is way too small for the profile and way too large for the material considering the amount of important information ellipsised in the profile in the main UI)

Yeah, good point. The width of the profile header is matching the drop-down beneath it, and maybe it shouldn't. But the story is the other way around for smaller screens. I think we need to make more things adaptive to the screen size.

> Create TPU Material for fun because to me it is just an easy way to change hotend/bed temperatures and retraction setting and it avoids me to create too much custom profiles

With this you also changed the Material Type then? That's still where it goes wrong and where the UX needs to be improved. If the PLA profile set should be used, the Material Type needs to be PLA.

fabriceimperial, 2020-10-02:
> No that's normal. Cura has two resource locations, one in its actual application and one in your `.local` folder, with identical folder structures. There will be definitions inside the `definitions` folder inside of the AppImage. The alternative location is in `.local` so that you could theoretically mod additional definitions there that you find on the internet or create yourself.

So maybe a blank template in the alternate folders could be useful (it is not uncommon on other software with profile based configurations to have default or readme file).

> Yeah, good point. The width of the profile header is matching the drop-down beneath it, and maybe it shouldn't. But the story is the other way around for smaller screens. I think we need to make more things adaptive to the screen size.

The possibility to be manually resizeable would be enough for me (like a default column header on grid components).

> With this you also changed the Material Type then? That's still where it goes wrong and where the UX needs to be improved. If the PLA profile set should be used, the Material Type needs to be PLA.

Yes, because my TPU is not 95 A in shore hardness but more like 80 A (actually 30 D).

gv2020, 2020-10-08:
created profiles in 4.7, upgraded to 4.7.1 was fine for a couple of days now all profiles in cura are gone (they are still present in appdata/....). also not able to create any new ones. Assume this relates to same bug

gv2020, 2020-10-08:
> created profiles in 4.7, upgraded to 4.7.1 was fine for a couple of days now all profiles in cura are gone (they are still present in appdata/....). also not able to create any new ones. Assume this relates to same bug.
> 
> profiles are in format mentioned above ([#8229 (comment)](https://github.com/Ultimaker/Cura/issues/8229#issuecomment-700348830))

may have discovered a non related bug, would be worthwhile verifying. i noticed i selected esun pla+ as material. when changing to generic PLA material they appear again.

Ghostkeeper, 2020-10-12:
Generic PLA and PLA+ are different material types according to the profiles we've received. They do not share their set of profiles.

printingotb, 2020-10-27:
I tried some of the things mentioned above with no success. Profiles would show within the profiles menu but not in the drop-down.

I managed to get my profiles working (from 4.3) by installing 4.8 and then remove contents of some of the appdata folders:
-definition changes
-machine instances
-materials
-quality changes
-user (may remember wrong on this one).

I then exported my profiles from 4.3 and imported them into 4.8 and they work now.
Hope this helps someone.

printingotb, 2020-11-03:
Creality machine-profiles have differently named quality types than most (all?) other third party machine profiles. This has changed somewhere between 4.3 and 4.8 beta, meaning that settings-profiles created on older Creality versions will not work with newer creality machine since these machines expect a different quality type. Settings-profiles created on newer Creality machines also won't be available on any other machine-profile (settings that I have used with different machine profiles, including Creality ones in the past).
They are created by Trouch.com. In Cura 4.3, I can't see the author.
[image]
[image]
[image]

spanzetta, 2020-11-23:
Hello, 
I still have the problem even after have tried all the suggestion here..
How is possible that after an upgrade an user cannot manage anymore its profiles?
Screenshots of my situation are shown in the closed ticket https://github.com/Ultimaker/Cura/issues/8818 
Can someone pls help?
Thanks

spanzetta, 2020-11-23:
also.. what is the exclamation mark shown here?

[image]

thanks in advance

fieldOfView, 2020-11-23:
The exclamation mark is a feature of the Sidebar GUI plugin that tells you that there are no profiles defined for the current material (PLA+) for the printer you have selected (Creality Ender 3 Pro).

spanzetta, 2020-11-23:
@fieldOfView thanks a lot for the explanation.. the reality is that I have 3 profiles (coming from previous Cura versions)..  but those are not shown and cannot be managed..
Did you see my screenshot?
Thanks

fieldOfView, 2020-11-23:
I meant standard quality profiles, not user-generated profiles. The warning icons shows that the printer definition says there are quality profiles supplied per material, but there are no quality profiles supplied with Cura for the material/printer combination.

Yes, I saw your screenshot with the arrow pointing to the warning icon that I put there, to warn you there are no quality profiles for the material/printer combination so the material/printer combination is "not supported" by the printer definition you selected. You should still be able to set settings, print and create user profiles ("quality changes" profiles in official Cura terminology). Afterall, it is a warning icon, not an error icon.

spanzetta, 2020-11-24:
I was referring to the screenshot in my other ticket .. 
here https://github.com/Ultimaker/Cura/issues/8818

devekrehbiel, 2020-11-24:
I have been out of commission for awhile and upgraded my Cura 3.3 to 4.8. Now I get:
Configuration errors
Your configuration seems to be corrupt. Something seems to be wrong with the following profiles:
-fdmextruder#2
-Anet_A8_AnetStd
-Custom FDM printer
-custom_my_cura_profile
Would you like to reset to factory defaults? Reset will remove all of your current printers and profiles.

So no, I want to get my Anet profiles working as usual. I read all of the above, but still not sure how to proceed.

Ghostkeeper, 2020-11-26:
Your issue is different from what this thread is about, Devekrehbiel. It's not polite to muddle the discussion here with unrelated configuration errors.

spanzetta, 2020-11-26:
dear @Ghostkeeper can you help me to understand how to solve my issue?
I found no way to recover my profiles.. so I am using Cura 4.8 with current profile and cannot make any changes.. 
Thanks in advance

Ghostkeeper, 2020-11-27:
Which profile was your custom profile based on? Is the profile available for different materials, like generic PLA?

tarsonis123, 2020-12-02:
Stumbled across this today. I tested an 0.2 nozzle and of course switched the nozzle size in the material tab. After a while I was looking for a preset and it was gone. Setting it to 0.4 made it visible again.  Totally intuitive such behaviour...
To be honest I massive dislike the general profile handling in cura. Usability wise a nightmare. For instance I the visibly settings that change every project one opens. I use the "custom visibility set" plug-in to cure this issue. Then this penetrant asking for changes on every project I want to open. And why the heck I can't select "Keep actual settings" in this dialogue. I don't want dozens of printers created for each project nore do I want dozens of the same material. Who wants this? The G-start code should be logically permanent/globally and for those special case project wise use a text field input option in the settings tab. I know one can change the behaviour in the settings but again where is "keep actual". There is discard. And there is update which mixes the actual settings with the one from the opening project. In this regard the preset management and usability are miles away from another.
I would like to not hack around with the presets just because a nozzle switch or have a text file open all the time with the actual start code to replace it with one of the project.

[39 more replies]

closed 2024-12-07 by github-actions[bot]