Ultimaker/Cura #8230: [4.7] Can not save custom start gcode if it contains an equal sign

mrzottel, 2020-08-25:
**Application version**
4.7

**Platform**
Windows 10 2004 64bit 

**Printer**
Ender 5

**Reproduction steps**
1. Enter custom gcode with an equal sign into custom start gcode
2. Close Cura 
3. Open Cura again 
4. Gcode is cut at the equal sign, ui falls apart (extruder settings missing, no materials, crash when loading stl)

**Screenshot(s)**
Inserting this gcode:
[image]

Results in this:
[image]

**Actual results**
Cura is unusable and crashes on loading a stl.

**Expected results**
Custom gcode containing equal sign should work. It is required by Octolapse in this case. 

**Additional information**
To reproduce this, I tried saving without equal signs - everything works (but Octolapse of course). I also checked the configuration files, and saving seems ok (I attached the file below). I think the issue comes from loading and parsing those config settings, as the equal sign is part of the configuration schema. 

After digging around the code and commits a bit - maybe the new FastConfigParser has an issue with this? (https://github.com/Ultimaker/Uranium/pull/625)

[Creality+Ender-5_settings.inst.cfg.zip](https://github.com/Ultimaker/Cura/files/5124194/Creality%2BEnder-5_settings.inst.cfg.zip)

fieldOfView, 2020-08-25:
Issue is likely related to the new [FastConfigParser](https://github.com/Ultimaker/Uranium/blob/master/UM/FastConfigParser.py)

fieldOfView, 2020-08-25:
It would have been nice if this were discovered during the beta phase.

mrzottel, 2020-08-25:
Sorry, I downloaded beta and found this two hours before the release - and then confirmed it in release as well =(

nallath, 2020-08-26:
Yeah. Now that I see the g-code, it's pretty clear that the fast config parser caused this. I didn't know that stuff like this put in the start gcode. 

Is it because the beta period was too short? I'm noticing that we're getting less bug reports from the beta, only to recieve them on the first "real" release (hence the high number of patch releases we had to do the past few releases).

mrzottel, 2020-08-26:
Well, for me it was just a coincidence. I was preparing a 2-day print and tried tree supports, but they generated partly in the air (well, found out what infill for supports is for then) so I checked if a new version of Cura is around. I saw the beta, got it, found the bug, wanted to post it, saw that the release is there, got the release, checked again - and well, here we are ;)

I just got into 3D-printing a month ago, but I think I will keep a closer look on upcoming betas to get into testing earlier ;)

nallath, 2020-08-26:
> I just got into 3D-printing a month ago, but I think I will keep a closer look on upcoming betas to get into testing earlier ;)

No problem. I'm just trying to figure out if there is anything we need to change in order to catch issues like this. Thanks for thinking along :)

Ghostkeeper, 2020-08-27:
We have a fix for this issue (see [the PR](https://github.com/Ultimaker/Uranium/pull/633)). We hope to release it somewhere next week. Thanks for the reports!

I'm amazed that you figured out that the `=` sign was at fault.

mrzottel, 2020-08-27:
Beeing a software developer for 20 years now I messed up so many things I became quite good at analysing that stuff (even though I never did anything with Python yet). 

I tested the fix and can confirm it works for me now. Thanks a lot!

closed 2020-08-27 by Ghostkeeper

ALSFelix, 2020-08-29:
It is not solved!
Just downloaded 4.7 and i do not have equal signs but it still loses all but the first 3 lines in extruder start code.

mrzottel, 2020-08-29:
I would suggest opening a new issue then. This issue is about the equal sign in gcode, which has been resolved and will be available in an upcoming release. Please attach the gcode to this new issue / fill out the template.

ALSFelix, 2020-08-29:
How do i go about openning a new issue?
Can you tell me?

ALSFelix, 2020-08-29:
Never mind... Just found it... Thanks