Ultimaker/Cura #8233: Upgrading to 4.7 = losing all presets. Why no real upgrade path?

globules-io, 2020-08-25:
**Application version**
4.7

**Platform**
Win 10 x64

**Reproduction steps**
1. Install 4.6
2. Setup your materials and settings
3. "Upgrade" to 4.7
4. Everything gone

**Expected results**
Cura should be able to update itself without losing presets. Upgrade path is unclear. Should not ask to uninstall previous but simply upgrade itself!

fieldOfView, 2020-08-25:
When you install a new version of Cura, the installer asks you (for each older version it finds) if you want to uninstall older versions. For each older version it uninstalls, it asks you if you want to remove all settings data.

globules-io, 2020-08-25:
And that does not imply my data.

fieldOfView, 2020-08-25:
Settings are no data?

globules-io, 2020-08-25:
And UPGRADING does not mean download full package, fully install and install new version in a new folder.
That's just .... installing a new version.
Why would I lose my data when upgrading any software? Why would I keep a previous version installed when I chose to UPGRADE.
Also, software settings can change from one version to the other. Many software ask to keep these settings (THE ONE CAN COME WITH THE SOFTWARE).

tslater2006, 2020-08-25:
When you do tell it to leave your presets/profiles you get this when you open 4.7
[image]
Which wants to delete all of the stuff anyways.

Liger0, 2020-08-25:
The possibility of having different cura versions at one is golden.

I upgraded my cura with creality machines with the pattern of:
4.6.2- > 4.7 beta -> 4.7
always selecting to uninstall the older version but keeping settings and no profile is damaged on my side.

nallath, 2020-08-26:
> Also, software settings can change from one version to the other. Many software ask to keep these settings (THE ONE CAN COME WITH THE SOFTWARE - NOT THE ONES I CREATE WHICH AS USUALLY STORED IN MY PERSONAL FOLDERS AND NOT THE PROGRAM FOLDER).

That is _exactly_ what Cura does. It asks you if you want to uninstall the previous version and if it wants you to delete any and all old data. It's a bit like with uninstalling a game and it asks you to delete your save data. If you press yes, it will simply do ask you asked for and well, delete the data.  

I understand it's very frustrating that you accidentally deleted all your data, but the installer is quite literally following the default configuration for installers.

nallath, 2020-08-26:
> When you do tell it to leave your presets/profiles you get this when you open 4.7
> [image]
> Which wants to delete all of the stuff anyways.

That is a different issue. Could you please create a new issue for that and fill in the bug template?

makemerush, 2020-08-26:
> > When you do tell it to leave your presets/profiles you get this when you open 4.7
> > [image]
> > Which wants to delete all of the stuff anyways.
> 
> That is a different issue. Could you please create a new issue for that and fill in the bug template?

I had the same issue. Previous Cura updates have not destroyed all of my previous custom printer profiles and settings but this upgrade to 4.7 does, so it appears to be corrupting them during the upgrade.

nallath, 2020-08-27:
> I had the same issue. Previous Cura updates have not destroyed all of my previous custom printer profiles and settings but this upgrade to 4.7 does, so it appears to be corrupting them during the [...]

**Please** fill in the bug template. If you don't, we can't do anything for you.

Ghostkeeper, 2020-08-27:
So this seems to be a misunderstanding of what the "Do you want to remove all your configuration files?" pop-up does in the uninstaller.

> And that does not imply my data.

What data would you expect it to delete then?

fieldOfView, 2020-08-27:
Since there are so many reports of people making the wrong decision here, it could be a good idea to make the default NOT to delete the configuration files.

Ghostkeeper, 2020-08-27:
If I read the NSIS template correctly, it's a pop-up message, not a checkbox with a default.
I'd need to start up a Windows computer to verify my claim though. Maybe I'm misreading it.

fieldOfView, 2020-08-27:
You are right, it is not a checkbox. The "Yes" (nuke my configuration!) choice is highlighted by default. The safer option would be to have the "No" (don't hurt my files) choice be the default. 

Many people just press enter on all dialogs when installing applications, without reading. It is often recommended in styleguides to have the default option be the safer option.

globules-io, 2020-08-27:
@Ghostkeeper 
> What data would you expect it to delete then?

Everything related to display preferences, window sizes etc. Everything about the program that is a setting already there when I install the program. Everything that I add as personal setting that does not exist once the program is first installed, belongs to the user and should be saved in his documents outside of the program folder.

Ghostkeeper, 2020-08-27:
But the display preferences, window sizes, etc. also don't exist yet once the profile is first installed. There is indeed a default for it, if that's what you mean, but there's also a default for the Infill Density setting of course.

We could consider making it reset just the preferences and not the slicing settings. But I don't really see a use case for someone wanting to reset their preferences but not the slicing settings. The main use cases for removing all configuration data is to leave a cleaner system and to revert to factory defaults, but neither of these are achieved if you leave the slicing settings intact.

fieldOfView, 2020-08-27:
The wording of the dialog could be changed to emphasize that removing the configuration data removes ALL CUSTOM PROFILES and CANNOT BE UNDONE possibly in all caps because apparently it needs to be spoken loudly for people to notice it.

globules-io, 2020-08-27:
@Ghostkeeper  the issue is rather simple, you are trying to complicate it, so let me rephrase one more time to make it even simpler.
Everything created from "create profile from current settings" is user data and should be saved outside the program folder like in documents. Rather simple no?

nallath, 2020-08-27:
I think that "flipping" the decision and improve the wording is a good way to do it. 

Not giving any option to delete those profiles is also a bit of a weird one, since these folders can get pretty big (especially if you download a lot of plugins). I would expect that a program deletes all the data it generated without me explicitly saving it somewhere

iMartyn, 2021-06-25:
I would actually suggest that if cura absolutely must do an uninstall and reinstall to upgrade (which is unusual to say the least), pass a parameter to the uninstaller that says no for you automatically.  I just lost all my settings with one click of a button during an upgrade because something distracted me at the wrong time.  I don't think anyone expects an upgrade to loose all their settings, and especially when slicing settings are such a big part of getting a 3d printer working, it should be as non-destructive as possible.

Ghostkeeper, 2021-07-02:
Cura doesn't need to uninstall at all. You can have multiple installations of Cura side by side. It's just that many users like to keep a clean computer, with just a single installation of Cura on it.
But most users won't be bothered by the few megabytes of logs and config that Cura keeps there, so making it a checkbox that is unchecked by default is still a good solution, I think.

fvrmr, 2021-07-15:
I have created a ticket on our backlog for this since it is unclear for a lot of users. 
Devs see CURA-8390

gman-bytes, 2022-02-08:
The update/upgrade path is NOT clear enough. I just lost configuration for 10 different printers with 3 to 5 preference variations on each printer. Total BS. Oh I should be more careful... etc etc. I also expect modern software to be somewhat bullet / fool proof.

gman-bytes, 2022-02-08:
Well they didnt get it done by version 4.13.1    I just lost a lot of work product.
So Its time to figure out who currently provides the best software for 3d printing world. Any real suggestions are welcome.
The cost of my error is in the $1000s so I am willing to pay for something better.

nallath, 2022-02-09:
> The cost of my error is in the $1000s so I am willing to pay for something better.

We recently had the first bug bounty of 500$. That issue was solved within a few hours ;) I understand why this issue is frustrating, and it is on our backlog to fix, but there are also other issues that also require our attention.

closed 2024-12-07 by github-actions[bot]