Ultimaker/Cura #8234: new Tevo Tarantula Pro profile has the bed 3mf vertically rotated 90 degrees

pabloandresm, 2020-08-25:
**Application version**
Cura 4.7

**Platform**
Windows 10 64 bit

**Printer**
Tevo Tarantula Pro

When you create a Tevo Tarantula Pro printer, the bed (tevo_tarantula_pro_platform.3mf) is badly shown vertically in the middle of the bed, instead of being horizontal and aligned.

To reproduce just create a new printer TevoTarantulaPro and see.

I've rotated and aligned the file if you are interested.

Here it is:

[tevo_tarantula_pro_platform.zip](https://github.com/Ultimaker/Cura/files/5125185/tevo_tarantula_pro_platform.zip)

Ghostkeeper, 2020-08-27:
Indeed! I wonder why we hadn't seen this when it was last updated. Maybe it's a relic of the conversion to 3MF to reduce file size.

You've made it very easy for us to correct. That's a drop-in replacement. Thank you!

Ghostkeeper, 2020-08-27:
I've committed it under your name: e13c48a8e76d4f52cea9636cfba2416f5b48084a
Congratulations on becoming a contributor to our source code :stuck_out_tongue_winking_eye:

closed 2020-08-27 by Ghostkeeper

pabloandresm, 2020-08-27:
Glad I could help! thank you very much

commit e13c48a8 (2020-08-27): Rotate and align Tevo Tarantula Pro build plate mesh

Fixes #8234.