Ultimaker/Cura #8235: Unable to open any STL in 4.7

tslater2006, 2020-08-25:
**Application version**
4.7

**Platform**
Windows 10

**Printer**
Creality Ender 5

**Reproduction steps**
1. Open Cura 4.7
2. Open STL file

**Screenshot(s)**
[image]

**Actual results**
Cura crashes

**Expected results**
It to not crash and to load the STL for printing.

**Project file**
N/A

**Log file**
I've submitted via the "Send Report"

mrzottel, 2020-08-25:
Any chance you are using octolapse and have the octolapse script in the starting gcode of your printer?

tslater2006, 2020-08-25:
I sure do.

[image: image.png]

mrzottel, 2020-08-25:
I added another issue with cura 4.7 not beeing able to cope with "=" signs in custom gcode, as used by this script. My workaround for now (aside from going back to 4.6.1 I used before) is either loosing the script for now and configuring octolapse by hand before each print or disable octolapse.

Unfortunately, I could only get cura 4.7 into an usable state again by removing all configuration from %appdata%/cura/4.7 and redo all my config.

fieldOfView, 2020-08-25:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/8230

fieldOfView, 2020-08-25:
> Unfortunately, I could only get cura 4.7 into an usable state again by removing all configuration from %appdata%/cura/4.7 and redo all my config.

That is somewhat unfortunate. I could have helped you remove just the offending =-signs in a text editor.
In the `definition_changes` folder, there are a couple of .inst.cfg files that you can open in a text editor to edit the custom start gcode you entered in Cura.

tslater2006, 2020-08-25:
I'm not convinced its a duplicate? the error message is referring to USB printer stuff, and this is when opening the model, not slicing it, why would whatever is in the start gcode (as referenced in #8230) have an effect at the point of just opening the STL.

mrzottel, 2020-08-25:
> That is somewhat unfortunate. I could have helped you remove just the offending =-signs in a text editor.
> In the `definition_changes` folder, there are a couple of .inst.cfg files that you can open [...]

Don't worry, it was not that much configuration, and most of it came from 4.6 backups and external text files. But I also tried to remove the = signs during testing, but that did not help much. It recovered the printer settings, but there seem to be follow up issues introduced (all materials were gone, which most likely causes the list index out of range issue in the crash). I guess this comes from loading and parsing the configuration with = signs and then saving the "broken" configuration again (opening and closing cura without loading a stl and causing the crash)

starlord1980, 2020-08-25:
Same problem here on Mac:
[image]

fieldOfView, 2020-08-26:
> I'm not convinced its a duplicate?

I have changed the wording above from "seems to be a duplicate"  to "could be a duplicate". However, the issue I linked to does also mention the crash when loading an STL. So I am fairly certain that it is indeed the same issue, and that it can be fixed by editing the files in `definition_changes` like I said above.

Ghostkeeper, 2020-08-27:
We have a fix for this issue (see [the PR](https://github.com/Ultimaker/Uranium/pull/633)). We hope to release it somewhere next week. Thanks for the reports!

It turns out that this had the same root cause as https://github.com/Ultimaker/Cura/issues/8230, although the symptoms were different.

closed 2020-08-27 by Ghostkeeper

Merkya87, 2020-08-27:
Hello @Ghostkeeper I am having the exact same issue can i just pull the updates somehow now before waiting for the fix to be released?

Thanks in advance!

fieldOfView, 2020-08-27:
That depends how you installed Cura and what OS you are running it on.

Merkya87, 2020-08-27:
> That depends how you installed Cura and what OS you are running it on.

I had Cura 4.6 previously, I have windows 10. I removed my old 4.6 version but kept the configurations.

fieldOfView, 2020-08-27:
It is not *very* easily hackable, but it is possible. The following instructions come with no warranty, perform at your own risk.

* Download this file: https://raw.githubusercontent.com/Ultimaker/Uranium/4.7/UM/FastConfigParser.py
* From `C:\Program Files\Ultimaker Cura 4.7\` copy `python35.zip` to your desktop
* Open `python35.zip` in a decent zip application (eg 7zip)
* Navigate to the folder `UM` and remove the file `FastConfigParser.pyc`
* In its place, put the file you downloaded in the first step. Note that it is named `.py`, not `.pyc`. Keep it that way
* Close the zip application, and copy the zip back into the `C:\Program Files\Ultimaker Cura 4.7\` folder

That should do it. Or it could blow up your kitchen. Or something inbetween.

Merkya87, 2020-08-27:
> It is not _very_ easily hackable, but it is possible. The following instructions come with no warranty, perform at your own risk.
>
> * Download this file: [...]

Awesome it works now thanks so much!

fieldOfView, 2020-08-27:
Did you check your kitchen?

mrzottel, 2020-08-27:
I repeated those steps and can confirm it works now. Thanks a lot!

Merkya87, 2020-08-27:
> Did you check your kitchen?

Kitchen is fine, my Ender 3 Pro is running good :)

mrzottel, 2020-08-27:
> Did you check your kitchen?

Mine is a wasteland - but thats caused by an uncatched exception in wife.dll ;)

starlord1980, 2020-08-28:
Doesnt seem to work on macOS. Replacing this file in:
/Applications/Ultimaker\ Cura.app/Contents/Resources/lib/python3.5/UM/FastConfigParser.pyc
just causes it to fail to launch. Restored the original from backup and it runs again.

I also noticed the linked file above is text, wheras the one in the Cura app package is not. Probably has something to do with it?

mrzottel, 2020-08-28:
.pyc is "compiled" python, while .py is the pure source code. 

If you just replace the FastConfigParser.pyc with FastConfigParser.py make sure, that you do not accidently rename FastConfigParser.py to FastConfigParser.pyc, because in this case, Python expects compiled code but gets uncompiled one.

craig-dae, 2020-08-29:
@Merkya87's solution confirmed to work for me

PDServices, 2020-08-30:
Works for me too, thanks

derekslenk, 2020-08-31:
I have this issue, but none of my start g-code even ported over so I don't have any remove to even test

Ghostkeeper, 2020-09-01:
> I have this issue, but none of my start g-code even ported over so I don't have any remove to even test

Try changing it to anything. Due to the same bug, the start g-code itself also gets cut off. Then the cut off part is reinterpreted as something else which causes the crash. If you change it, it should re-save the profile and hopefully fix the problem.

derekslenk, 2020-09-01:
> Try changing it to anything. Due to the same bug, the start g-code itself also gets cut off. Then the cut off part is reinterpreted as something else which causes the crash. If you change it, it [...]

I had to delete my old 4.6 folder from %APPDATA%\cura to get anything to work. I tried deleting the start code but that didn't fix it. I also uninstalled 4.7, removing preferences, and reinstalling didn't fix the issue without deleting the old APPDATA first.