Ultimaker/Cura #8236: GUI glitch when scrolling [Version 4.7]

mbmcnicol, 2020-08-25:
**Application version**
Version 4.7

**Platform**
Windows 10 64-bit

**Printer**
Ender 3

**Reproduction steps**
1. Open list of print settings.
2. Scroll.

**Screenshot(s)**
[image]

**Actual results**
While scrolling, numbers will disappear and reappear.  If you release the mouse button while the numbers are gone, they will remain gone until either (1) you continue scrolling or (2) the mouse cursor enters a box where the numbers should be.  The values are retained, this appears to be a graphical glitch only.

**Expected results**
Numbers should not disappear; they should always be visible.  This was the behavior in Version 4.6.  I uninstalled Version 4.6, retained my user files and settings, and installed Version 4.7 and saw this behavior start.

**Project file**
N/A, not related to a project.

**Log file**
N/A, no log file will be created in duplicating this issue.

**Additional information**
No additional, relevant information is available.

Liger0, 2020-08-26:
I also have this issue since 4.7 beta.

nallath, 2020-08-26:
> I also have this issue since 4.7 beta.

Then why didn't you report it? It's now very unlikely to be fixed for 4.7.

smartavionics, 2020-08-26:
I think it's also in the current master branch...

nallath, 2020-08-26:
> I think it's also in the current master branch...

Stands to reason, since 4.7 isn't that much different from master at the moment. But it must have something to do with drivers / setup, since not everyone has the issue.

Ghostkeeper, 2020-08-27:
I could reproduce this issue but only in the (AppImage) build, not when running from source. Here is a recording of how it looks in animation:

http://dulek.net/work/setting_list_glitchy.webm

Pretty much as described. But three observations:
* Like 90% of the time it's okay. In this video I purposely stopped scrolling while it was gone.
* It never happens when scrolling up, for some reason.
* It doesn't affect some settings. Which settings it affects seems to depend on their position on the screen.

Barry37, 2020-09-02:
It does this on Mac OS X 10.10 Yosemite, too.
Mostly, but not always, scrolling towards the bottom of Settings (Experimental).
More likely to occur when scrolling with mouse pointer over the scrollbar, or dragging the scrollbar.

ivanmalvin, 2020-09-09:
Same issue here. Windows 10, Cura 4.7 and still present on 4.71. Scrolling down the Print Settings causes the values to disappear or flicker. Doesn't seem to happen when scrolling up. Never had the issue before in Cura 4.5, 4.3, 4.1, and earlier versions etc--seems new in 4.7 and 4.71.

nallath, 2020-09-09:
I did a bit of investigation into this and it makes absolutely _no_ sense that it's doing it. If I change the font of the text box to anything but what it is now, the bug disappears. It's probably not what you want to hear from a developer, but I honestly have no idea what the hell caused this.

Ghostkeeper, 2020-09-10:
Probably running into the same problem as this guy: https://forum.qt.io/topic/54693/listview-text-disappears-when-scrolling-delegate-is-text-over-image/5
But there is no solution there either.

Barry37, 2020-09-10:
> I did a bit of investigation into this and it makes absolutely _no_ sense that it's doing it. If I change the font of the text box to anything but what it is now, the bug disappears. It's probably not what you want to hear from a developer, but I honestly have no idea what the hell caused this.

This bug doesn't change if the "weight" settings in theme.json are altered - I've been using an edited version since the "Extra Bold" look started.
I've gone back to 4.6.1, as the Layer View in 4.7 is patchy (literally). I'm using Yosemite (OS X 10.10.5), which may the reason for layers not displaying correctly, as the Cura version 4.7.1 seems to be for OS X 10.11 and higher.

Liger0, 2020-09-10:
Wouldn't it be possible to compare the change commits between 4.6.2 and 4.7 beta and adding them some at time to see which commit caused this?

Ghostkeeper, 2020-09-11:
> I'm using Yosemite (OS X 10.10.5), which may the reason for layers not displaying correctly, as the Cura version 4.7.1 seems to be for OS X 10.11 and higher.

Since Cura 3.4 we're using Qt 5.10, which supports down to [MacOS 10.11](https://doc.qt.io/archives/qt-5.10/supported-platforms-and-configurations.html), not Yosemite. Our build server is also on 10.11 so CuraEngine has the same requirement. I don't know what could be going wrong in unsupported old versions but since Qt is involved it would make sense that the rendering is buggy. This hasn't changed since Cura 3.4.0 though.

> Wouldn't it be possible to compare the change commits between 4.6.2 and 4.7 beta and adding them some at time to see which commit caused this?

Yeah that would make sense. However the commits that could influence this are spread out between the Cura and the Uranium repositories which makes a normal `git bisect` a bit more involved. This is how I would approach tracing down bugs like this though.
There are also [1134 commits](https://github.com/Ultimaker/Cura/compare/4.6...4.7) between 4.6 and 4.7 on the Cura repository alone, so it's no small task in any case.

Liger0, 2020-09-11:
So you can't only select the minor versions like 4.6.2 vs 4.7 beta rather than 4.6 to 4.7? Pretty bad. :\

Barry37, 2020-09-12:
> Since Cura 3.4 we're using Qt 5.10, which supports down to [MacOS 10.11](https://doc.qt.io/archives/qt-5.10/supported-platforms-and-configurations.html), not Yosemite. Our build server is also on [...]
That probably explains it, though layer view was OK until 4.7. Oddly, most of each layer's missing render appears if "Helpers" is checked, even though that seems to mean show Brim or Skirt.

Ghostkeeper, 2020-09-15:
> So you can't only select the minor versions like 4.6.2 vs 4.7 beta rather than 4.6 to 4.7? Pretty bad. :\

Oh you can. There are [1030 commits](https://github.com/Ultimaker/Cura/compare/4.6...4.7-beta) from 4.6.2 to the beta.

BlondieGayMan, 2020-10-12:
I held off installing 4.7.1 and was using 4.3 for a long time.  But I finally installed it this week.  Sure enough, the flashing text when scrolling down (but not scrolling up).  It's not a deal-breaking problem, just annoying.  I just thought I'd confirm.

BlondieGayMan, 2021-01-06:
So here we are in Jan2021.  The problem is also there in version 4.8!  I've confirmed it.
This is NOT a "driver" issue!  This is a CODING issue!  Just because SOME people aren't affected does not mean it's driver related.  I've been down this exact road before with one of the products a company I worked for designed.
We had only some people that had a specific problem that sure appeared to be "graphics driver" related.
In fact, as a member of the team, I got into some tussles with the leads who where convinced it was people's drivers causing the issue.

I was able to duplicate the bug on my lab computer.
The team lead came to this machine and created break points in the software.
When I was able to re-create the bug, I still remember his words,  he said, "hmmm.... this SHOULD affect everyone!"
He saw the issue right away and it was a very quick and simple fix IN THE CODE!

When I'm coding I don't make TOO MANY errors (cough).... but when I do, it's also usually something simple and a typo or missing character.

The problem here is that this problem appears to be falling on deaf ears.  OR, perhaps the devs don't consider this an urgent issue.  I don't know.
But a bug in the code it most certainly is.
The fact that when you scroll UP, the issue does not happen but when you scroll DOWN it happens, you'd think that THIS should make it easier to located and address this bug.

One who used to be a programmer from days gone by, would think it should be easy to locate the function/module that it's happening in.  Break points could be set up.

Oh well.  It's in 4.8 too and has not been addressed.

I've updated to 4.8 due to some features I wanted to use and like before, while this bug is annoying, it's not a deal breaker.

In fact, I'm very impressed with the coding the devs do to create such an amazing and full-featured slicer.

Ghostkeeper, 2021-01-18:
I don't think any of the developers here is under the impression any more that it's a driver issue. We've since reproduced the problem on multiple operating systems with completely different drivers (Windows and Linux). And we had a report for MacOS as well.

I also don't think this is the result of a typo though. Your anecdote is a bit moot since not all software bugs are typos. We are not making the visibility of this text depend on whether you are dragging the scrollbar, and if we did the text wouldn't reappear when you hover the mouse over it. I'm 95% convinced this is a bug in Qt with how text is rendered on fractional pixel coordinates. I don't think it's hard to workaround when we find the issue, but finding the issue is not a matter of "look where we scroll up" since the code that handles scrolling is not in Cura but in Qt.

That said, indeed the developers don't consider this to be an urgent issue considering the settings automatically reappear if you try to use them afterwards. It's also the type of problem that could automatically be fixed by upgrading our Qt version.

BlondieGayMan, 2021-01-20:
Thanks for confirming that.  As you mention that text is rendered on  fractional pixel coords, that now makes more sense.
Also, just for clarity, when I talk about a typo, that was only an example, coming from personal experience.  I wasn't stating that this must be what it is.  LOL
We all know most problems are easy to fix.  Like you say, locating it is the issue.

Anyway, hopefully one day the devs will consider it "urgent" as so many people are mentioning it.

I do understand however.  Been there, done that.  Even during dev, we know that each coder or team lead/manager will have differing opinions on the urgency of an issue. 

Thanks for reaching out and confirming.

I'll say this, that other than this annoying (but not critical) issue, 4.8 is turning out to be very stable for me.

Thanks so much and please do take care.

Ghostkeeper, 2021-01-28:
Also noticing this issue when scrolling all the way down, sometimes.

I strongly suspect this is an issue with rounding to whole pixel coordinates somewhere.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

BlondieGayMan, 2024-11-22:
I can't speak for others, but for me, this is no longer an issue as I bought 2, Ender 3 V3 Plus printers and no longer use the junk Wanhao.  

With the enders, I'm using Creality Print.

So for me, no issue any longer.

👍😁👍

closed 2024-11-22 by GregValiant

Barry37, 2024-11-23:
I’d forgotten about this one. Not sure which version it had stopped by, but it doesn’t do it in 5.8.
👍