Ultimaker/Cura #8237: Slicing never finishes.

Hix777, 2020-08-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Liger0, 2020-08-25:
The template, tho.

enkiera, 2020-08-26:
I am also getting the same issue after upgrading to 4.7. when trying to slice any model the slicing never completes. The loading bar on the bottom right doesn't show any progress. I've tried reinstalling and downgrading back to 4.6 but this is now also doing the same thing.

jellespijker, 2020-08-26:
please fill in the bug report and send us your log

closed 2020-09-09 by no-response[bot]