Ultimaker/Cura #8239: Import profiles impossible

loulou-uoluol, 2020-08-26:
**Application version**
4.7
**Platform**
Mac OS X ( Catalina)

**Printer**
Predator (Cura Printer)
Predator (my Custom printer)

**Reproduction steps**
1. export an profile (my custom printer)   
2. Import the profile (Predator)
3. Unknown file type or is corrupted.
4. Same message with olds profiles

**Screenshot(s)**
[image]

nallath, 2020-08-26:
Please fill in the entire bug template. We're missing log files and the profile files that don't work for you.

I've just tried to reproduce this, but the profile exported and imported  correctly.

loulou-uoluol, 2020-08-26:
I completely removed Cura, and after restarting my mac.
I reinstalled Cura. It works normally, I can import my profiles.
Thanks for your help.

closed 2020-08-26 by loulou-uoluol