Ultimaker/Cura #8240: Individual exclusive slicing tolerance creating empty first layers at 0

Asterchades, 2020-08-26:
**Application version**
4.7, unsure if it existed earlier

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 x64 Pro 2004
NVidia GeForce GTX 1080

**Printer**
(Which printer was selected in Cura?)
Custom profile based on FFF/FDM default.

**Reproduction steps**
1. Place 2 or more models on the print bed at the same time. They do not have to be the same model.
2a. Assign a global Slicing Tolerance of "Middle" or "Inclusive" (default Middle)
2b. Assign a global Slicing Tolerance of "Exclusive"
3a. Assign individual models a Slicing Tolerance of "Exclusive", so long as one is left at Middle/Inclusive
3b. Assign individual models a Slicing Tolerance of "Middle" or "Inclusive", so long as at least one is left Exclusive
4. Observe

(NB Only follow the A path or B path, since you can't follow both at once)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)
Any model which is still left set to an Exclusive Slicing Tolerance will not have a presence on the first layer while their Z location is set to 0, either manually or automatically. Lowering it by even the smallest of amounts (-0.001mm) will restore the first layer, though may naturally have (ever so slight) implications through the rest of the model.

In the event all models are set to Exclusive or none of them are, the first layer will print as normal.

**Expected results**
(What should happen after the above steps have been followed.)
No empty layers should be produced on any of the models, and slight movement should not be required to compensate.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[Profile.zip](https://github.com/Ultimaker/Cura/files/5128224/Profile.zip)

[Exclusive test.3mf.zip](https://github.com/Ultimaker/Cura/files/5128227/Exclusive.test.3mf.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
I suspect this is strictly an edge case, since I would doubt it will see much impact outside of tests which explicitly reveal it. Still, it may have other flow-on effects which aren't so easily observed so it may be worth investigating all the same.

smartavionics, 2020-08-26:
See https://github.com/Ultimaker/CuraEngine/pull/1306

Asterchades, 2020-08-26:
While likely technically accurate, my concern is that the difference is actually going to impede accuracy (thus tolerance) in this specific example, in addition to faulting one of the settings.

The fault, of course, is that I have the model set to a 0 Z location. This location is provided by the "Automatically drop models to the build plate" setting. By not printing the first layer it's effectively making this offset equal to my initial layer height, which is not what I have asked for nor what has been generated. I asked for the model on the bed - not floating above it.

The issue with accuracy is going to come from this missing first layer resulting in the second layer (the first of that particular mesh) now being stretched. Its going to droop, whether I like it or not (and whether Cura likes it or not), and that will have a flow-on effect for the rest of the model as each subsequent layer will not be assured of a stable foundation at the expected height. Plus it will negatively impact bed adhesion in the same way that having the incorrect Z level will, greatly increasing the likelihood of a failed print.

Given the above (and I realise I'm a little bit "preaching to the choir" given you started that PR) I can't help but feel that the ideal solution would be to compensate for this removal by shifting the affected model/s - without user intervention - down by a single layer, the same as it would if all models were set Exclusive. This satisfies the user request (place it on the bed) while satisfying the feature design (applying 1 layer of tolerance), but without the negative side-effects.

Interestingly, I didn't find similar (mis)behaviour with Middle vs Inclusive. But it may be more fine-grained than what I was looking for.

smartavionics, 2020-08-26:
Personally, I never use anything but middle tolerance as I feel that the other modes are not trustworthy either in concept or execution. If I want a part to fit, I alter the design, I don't expect the slicer to frig with it.

Asterchades, 2020-08-26:
Oh I agree. But not everything I print is going to be something I've designed, and altering STLs can be a massive pain in the butt depending upon how they were designed in the first place and what tools were used.

However I'm leaning more towards switching to Inclusive myself. Spheres or other rounded features, in my experience, tend to be missing the very top layer when set to Middle - my suspicions being that trying to find the middle between a known shape and the absence of any shape at all is either causing an issue or being deliberately discarded to avoid issue.

For example, try poor Phil-A-Ment, at 50% scale, on Cura, Slic3r, and Ideamaker. Cura prints one layer less than the other two even left at Middle, with what appears to be a single layer "dome" lopped off the top as compared to the object preview.

smartavionics, 2020-08-26:
> For example, try poor Phil-A-Ment, at 50% scale, on Cura, Slic3r, and Ideamaker. Cura prints one layer less than the other two even left at Middle, with what appears to be a single layer "dome" lopped off the top as compared to the object preview.

Sure, it's always possible to find examples that cause a problem. It should, however be possible to get a good result using middle by tweaking the layer height so that the model height is very slightly higher than a whole number of layers. Keeping the layer height constant (as we nearly always do) and then expecting the slicer to "do the right thing" WRT the topmost layer is kinda lazy. Hey, I'm not saying I don't do the same, I'm as lazy as everyone else in this regard.

Asterchades, 2020-08-26:
Lazy, perhaps, but it's not unreasonable. I guess this is one of those situations where "right" is subject to some interpretation, though. Hell, there's three different interpretations in just that setting. Just one of the many reasons why 3D printing is unlikely to go mainstream any time soon, I guess.

And I deliberately picked Phil because I don't consider it an edge case. The top of his head is, for all intents and purposes, a sphere. And with the number of people recommending simple programs like TinkerCad to create models, plus the existence of Paint3D, spheres are to be expected.

Ghostkeeper, 2020-08-28:
It's going wrong with edge cases here. That pull request at Ultimaker/CuraEngine#1306 makes a fix to the edge case at the first layer, by moving the slice position of that first layer up by 1 micrometre in this specific case. But the problem really happens with all layers, not just the first.

The edge case of that pull request is that Cura makes a slice at Z=0mm when using Exclusive or Inclusive tolerance. Since the model is aligned to Z=0mm precisely, there will be several triangles with two (or three) vertices at Z=0mm. [Here](https://github.com/Ultimaker/CuraEngine/blob/8971cfef7d91c5ca808dc76c58ce8e9a1adee24a/src/slicer.cpp#L851-L892) is where it registers which triangles intersect with a layer, and as you can see, it has several checks where it looks for `p0.z < z` and such. That means that it considers it an intersection if this vertex is strictly lower than the `z` at which we're making the cross section. So if a vertex is exactly on that cross section, the polygon is not considered to be intersecting.
On the other hand, at the high end we often see `p2.z >= z`, so there it does consider the triangle to be intersecting if this vertex is exactly on the Z coordinate of the cross section.

So when you have an object set to Exclusive and a different object (with e.g. Middle) besides it, it considers the cross section at Z=0mm and the cross section at Z=0.1mm. The cross section at Z=0mm is always empty because at best the vertices will be exactly at Z=0mm, never below it. So Exclusive will make the entire layer empty. Because there is something besides it that doesn't have that problem, the Remove Empty First Layers setting doesn't kick in and you get a floating model.

I have a [test model](https://github.com/Ghostkeeper/SlicerTestModels/blob/master/text/layer_numbers.scad) that reveals that the problem continues on the other layers too. Because for the second layer there is a similar edge case. If two vertices of a triangle lie exactly on a cross section and the third vertex goes up, it is not considered to be a part of the cross section. On the other hand, if the third vertex goes down, it is considered to be part of the cross section. So the total number of layers with content is then still correct, but you get the effect that all cross sections are shifted up by one layer. It considers the top side of the model even though it's an edge case there, but it doesn't consider the bottom side while it's really the same edge case.

I'll adjust the edge cases to remedy this.

closed 2020-09-03 by rburema