Ultimaker/Cura #8243: Add "Artillery Sidewinder X1" as printer

cataclism, 2020-08-26:
I'd like to have the preset ready of the Artillery Sidewinder X1, instead of having to set everithing everytime.
I have used the 3DNexus machine preset, that can be found [here](https://3d-nexus.com/resources/file-archives/download/8-artillery-evnovo/32-3dn-artillery-swx1-and-genius-cura-4-6-1-machines-and-profiles-installation-package)

All the Artillery community can get benefit from this addition, and it is large.

Ghostkeeper, 2020-08-28:
The developers of Cura can't add profiles to Cura for printers which we don't have. We can't test these profiles at all. So we rely on the community to test them and add them to Cura. This involves creating a pull request and then passing our tests and code review. The tests and review aren't very hard, but we do have some more restrictions than what you'd find on 3DNexus. For instance we limit the file size of build plate meshes to 1MB (we've gotten models of 100MB before) to limit the installation size. And we have checks that prevent unnecessary overrides, to improve performance and maintainability.

The profiles in question are also outdated and don't load in Cura 4.6.2 or 4.7.0.

If you have profiles that work in the current version of Cura, we'd be happy to accept them via a pull request! But we're not going to create them for you.

closed 2020-08-28 by Ghostkeeper

cataclism, 2020-08-28:
Ok, I'll create a pull request! Thanks @Ghostkeeper =)