Ultimaker/Cura #8245: Cura 4.7.0 resets after creating new profile

skarasov, 2020-08-26:
**Application version**
4.7.0

**Platform**
Win 10 2004

**Printer**
Anycubic i3 mega

**Reproduction steps**
1. Start Cura 4.7.0 (fresh install in my case).
2. Welcome screen -> skip cloud -> Add printer Anycubic i3 mega.
3. Goto prefs -> Profiles -> Duplicate "normal" profile, activate it.
4. Close all.
5. Start Cura - again.

**Screenshot(s)**
[image]

**Actual results**
Cura welcomes you again and error msg about profiles.

**Expected results**
Work without reseting.

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5130413/cura.zip)

nallath, 2020-08-26:
It specifically breaks for the duplicate. It doesn't happen when you use "create new profile from settings/overides"

choppy420, 2020-08-26:
I to am having a problem importing the profiles, after I import them the do not show up

nallath, 2020-08-27:
> I to am having a problem importing the profiles, after I import them the do not show up

Okay, could you create a new issue for this and fill in the bug template?

closed 2020-08-27 by Ghostkeeper

Ghostkeeper, 2020-08-28:
We've fixed this issue in an upcoming release, likely early next week.

commit 2b4d9f64 (2020-08-27): Ensure that quality_changes profile is correctly created on profile duplication

I have no idea why it wasn't being created, but it has been wrong for a very long time.

Fixes #8245
PR #8246 (2020-08-27): Ensure that quality_changes profile is correctly created on profile duplication
I have no idea why it wasn't being created, but it has been wrong for a very long time.

Fixes #8245