Ultimaker/Cura #8247: "outer before inner walls" affects bridge walls

osterac, 2020-08-26:
**Application version**
(The version of the application this issue occurs with.)
4.7
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Pro v 2004
**Printer**
(Which printer was selected in Cura?)
Creality Ender 4 (with coreXY mod)
**Reproduction steps**
1. (Something you did.)
Open the one layer bridging test model ([](https://www.thingiverse.com/thing:2614805))
turn outer before inner walls off
slice model
check preview at layer 16, the layer with the bridging
play the layer print preview
the first thing to be printed will be the overhanging bridge walls, from the furthest point away from the existing wall (in thin air) to the closest. 
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
Printing this slice will result in spaghetti.
**Expected results**
(What should happen after the above steps have been followed.)
slicer should tell the printer to print the 3 walls of the actual model first, inside to out, and then start on the bridge walls; what I guess would be outside to in?
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[BRIDGE_-_One-layer_bridging_test.zip](https://github.com/Ultimaker/Cura/files/5131284/BRIDGE_-_One-layer_bridging_test.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.zip](https://github.com/Ultimaker/Cura/files/5131296/cura.zip)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-08-26:
Arguably, when an inner wall is completely (or at least mostly) above air, it should be discarded and the skin expanded to fill the gap. I haven't looked at the detail of this but I suspect that it would be a lot of work to achieve. I will take a look though to see if it is possible.

osterac, 2020-08-26:
I updated the post with this, FYI.
with "outer before inner walls" turned on, bridge walls print in the correct (opposite) order.

If what you're talking about is a lot of work, wouldn't it be easier to have the "outer before inner walls" setting not affect bridge walls? Also, cura 4.7 is printing this with 3 bridge walls. 4.6.1 printed it with 2. Is this intentional?

smartavionics, 2020-08-26:
> If what you're talking about is a lot of work, wouldn't it be easier to have the "outer before inner walls" setting not affect bridge walls?

TBH, I don't think that printing any totally unsupported walls  is worth doing, irrespective of the order in which they are printed. I would rather put time into a solution that could yield a good result.

> Also, cura 4.7 is printing this with 3 bridge walls. 4.6.1 printed it with 2. Is this intentional?

I don't know if that is intentional or not. I see that the centreline of the 2nd green line is in the unsupported area so it doesn't surprise me that it is considered a bridge line.

osterac, 2020-08-26:
> TBH, I don't think that printing any totally unsupported walls is worth doing, irrespective of the 
order in which they are printed. I would rather put time into a solution that could yield a good result.

Well it's cura that is placing the unsupported walls, yes? Wouldn't that be a matter of changing the way cura bridges gaps? It sounds like you are suggesting doing away with bridge walls entirely.

smartavionics, 2020-08-26:
As I understand it, you're saying that when the inner walls are bridges then the print outer wall first setting should be ignored and that the walls should always be printed from the outer wall inwards.

What I'm saying is that in the situation where the inner walls are totally unsupported (as in your example above), those totally unsupported inner walls could be removed and the skin expanded to fill the space those walls would have filled. It's not obvious whether that would be worth doing or not.

Maybe this is all a waste of time as bridging is not really aimed at being able to print totally unsupported walls but rather spanning gaps.

osterac, 2020-08-26:
If it helps, this is what the layer looks like when finished
[image]

smartavionics, 2020-08-26:
I think we are on different wavelengths...

osterac, 2020-08-26:
yes...
so this is with "outer before inner walls" on:
[image]

This is off:
[image]

osterac, 2020-08-27:
> As I understand it, you're saying that when the inner walls are bridges then the print outer wall first setting should be ignored and that the walls should always be printed from the outer wall inwards.

Yes, that's it. Sorry, I missed this.

> What I'm saying is that in the situation where the inner walls are totally unsupported (as in your example above), those totally unsupported inner walls could be removed and the skin expanded to fill the space those walls would have filled. It's not obvious whether that would be worth doing or not.

What's confusing me is that it seems you are talking about correcting a bug by introducing a new feature rather than going after the source of the bug? What other situations would this new feature apply to?

Sorry if I've been a bit obtuse.

smartavionics, 2020-08-27:
> What's confusing me is that it seems you are talking about correcting a bug by introducing a new feature rather than going after the source of the bug? What other situations would this new feature apply to?

I don't see this as a bug in Cura, True, it could do more work, and one possible strategy could be to automatically remove the walls that are totally unsupported and replace them by skin but even doing that doesn't guarantee a good result. Even so,  you can still (possibly) avoid this issue by either not using 4 walls or by printing the outer wall first. TBH, I think you are somewhat optimistic in expecting a good result when printing unsupported walls. That's all from me, bye!

osterac, 2020-08-28:
Maybe I'm getting my terminology confused? Isn't the "skin" of the brigde the parallel horizontal lines that go over the gap and the "bridge walls" are just an extension of the perimeters surrounding the gap? If so, I can't avoid the issue by not using 4 walls (or is it 3?) because cura decides that and gives me no option to do otherwise.

This is a test print I and others use to tune my bridging settings. I'm not expecting good results, just the best I can manage.

viper0078, 2020-08-28:
Excuse me for cutting in.
I don't think it's a bug either.
Regarding the "Outer Before Inner Walls" parameter, how should the slicer behave if there is a mix of parts that should be on and off in the same printout?
It would be ideal if everything could be detected automatically to get the best possible printout, but I think it would be difficult because it depends on various conditions such as shape and support.
The same phenomenon occurs in Simplify3D.
I would need to check the slice results and add support.
I don't want the slicer to change its shape on its own in order to get the best printing results. Technically it could be done, but I don't think I would use it. Add support as appropriate to get the printout as designed.

Ghostkeeper, 2020-08-28:
> Also, cura 4.7 is printing this with 3 bridge walls. 4.6.1 printed it with 2. Is this intentional?

This could be the result of fixes we made w.r.t. line overlaps, maybe? We've made some fixes to how the insets are calculated with Infill Overlap, Skin Overlap, etc. It's hard to say.

I think Viper brings up an important point here, and I think SmartAvionics also saw that (through his wording of "totally unsupported"). We're looking at a very specific case here where the bridge is supported on all sides by a rim that's thinner than the walls. Only when the rim is thinner than the walls will the innermost walls be completely unsupported. For instance, here I've modelled something similar to what you've posted, but on one side the wall is wider:
[image]

Here it still goes from inside to outside as the disabled "Outer Before Inner Walls" setting requires. That's fine for the bottom right hand side of the stair, but it's going to hang in the air for the top side. Would you expect it to switch halfway through printing the contour? That would add additional seams though. Would you expect it to print the entire thing from outside in, maybe?
I can make it more extreme, too. It could be just a tiny strip where the wall is unsupported:
[image]

I consider this a bit of an exceptional case. In practice it will be very rare to encounter this, unless you have a construed case like that bridging test. Considering the difficulty to detect the case and perhaps to break up the contour planning, I don't think it's worth the time investment to fix this case.

osterac, 2020-08-28:
Upon further investigation, I think this may have more to do with the "alternate extra wall" setting. I had that setting on, and it was printing an extra bridge wall and printing them in the wrong order (the first wall would be in midair). With it off, they print in the right order with 2 instead of 3 bridge walls.

osterac, 2020-08-28:
So I made a simpler version of the test print, without all the rounded corners. This slices correctly:
[image]
I'll use this from now on. here's the file if anyone wants it

[Bridge Test Simple stairs.zip](https://github.com/Ultimaker/Cura/files/5144438/Bridge.Test.Simple.stairs.zip)
Thanks everyone for looking into this.

closed 2020-08-28 by osterac

Ghostkeeper, 2020-09-01:
Really, the difference is just that you reduced the wall thickness, so it doesn't matter that it starts from the inside because the inside is still above the edge of the model on the previous layer.